lsongdev/node-escpos #1: unable to install node-escpos

kamaldeepoodles, 2015-12-16:
I am trying to install node-escpos using followind command on windows 7 x32

npm install node-escpos --save

But unfortunately got bunch of errors....

D:\testndelete\testThird>npm install node-escpos --save

> iconv@2.1.11 install D:\testndelete\testThird\node_modules\iconv
> node-gyp rebuild

D:\testndelete\testThird\node_modules\iconv>if not defined npm_config_node_gyp (node "C:\Users\xz\AppData\Roaming\nvm\v0.12.7\node_modules\npm\bin\node-gyp-bin\\..\..\nod
e_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node  rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
Project file contains ToolsVersion="14.0". This toolset may be unknown or missing, in which case you may be able to resolve this by installing the appropriate version of
 MSBuild, or the build may have been forced to a particular ToolsVersion for policy reasons. Treating the project as if it had ToolsVersion="4.0". For more information,
please see http://go.microsoft.com/fwlink/?LinkId=291333.
C:\Program Files\MSBuild\Microsoft.Cpp\v4.0\Platforms\Win32\Microsoft.Cpp.Win32.Targets(511,5): error MSB8008: Specified platform toolset (v140) is not installed or inva
lid. Please make sure that a supported PlatformToolset value is selected. [D:\testndelete\testThird\node_modules\iconv\build\libiconv.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\xz\AppData\Roaming\nvm\v0.12.7\node_modules\npm\node_modules\node-gyp\lib\build.js:270:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "node" "C:\\Users\\xz\\AppData\\Roaming\\nvm\\v0.12.7\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd D:\testndelete\testThird\node_modules\iconv
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok
D:\testndelete\testThird
└─┬ serialport@2.0.5
  ├── async@0.9.0
  ├── bindings@1.2.1
  ├─┬ debug@2.2.0
  │ └── ms@0.7.1
  ├── nan@2.0.9
  ├─┬ node-pre-gyp@0.6.13
  │ ├─┬ npmlog@1.2.1
  │ │ └─┬ are-we-there-yet@1.0.4
  │ │   └─┬ readable-stream@1.1.13
  │ │     ├── inherits@2.0.1
  │ │     ├── isarray@0.0.1
  │ │     └── string_decoder@0.10.31
  │ ├─┬ request@2.65.0
  │ │ ├── aws-sign2@0.6.0
  │ │ ├─┬ bl@1.0.0
  │ │ │ └─┬ readable-stream@2.0.2
  │ │ │   ├── inherits@2.0.1
  │ │ │   ├── isarray@0.0.1
  │ │ │   ├── string_decoder@0.10.31
  │ │ │   └── util-deprecate@1.0.2
  │ │ ├── caseless@0.11.0
  │ │ ├─┬ combined-stream@1.0.5
  │ │ │ └── delayed-stream@1.0.0
  │ │ ├── extend@3.0.0
  │ │ ├── forever-agent@0.6.1
  │ │ ├── form-data@1.0.0-rc3
  │ │ ├─┬ har-validator@2.0.2
  │ │ │ ├─┬ chalk@1.1.1
  │ │ │ │ ├── ansi-styles@2.1.0
  │ │ │ │ ├── escape-string-regexp@1.0.3
  │ │ │ │ ├─┬ has-ansi@2.0.0
  │ │ │ │ │ └── ansi-regex@2.0.0
  │ │ │ │ ├─┬ strip-ansi@3.0.0
  │ │ │ │ │ └── ansi-regex@2.0.0
  │ │ │ │ └── supports-color@2.0.0
  │ │ │ ├─┬ commander@2.9.0
  │ │ │ │ └── graceful-readlink@1.0.1
  │ │ │ └─┬ is-my-json-valid@2.12.2
[... 109 lines cut ...]
npm ERR! Windows_NT 6.1.7601
npm ERR! argv "C:\\Program Files\\nodejs\\\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "node-escpos" "--save"
npm ERR! node v0.12.7
npm ERR! npm  v3.4.0
npm ERR! code ELIFECYCLE

npm ERR! iconv@2.1.11 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the iconv@2.1.11 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the iconv package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls iconv
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     D:\testndelete\testThird\npm-debug.log

D:\testndelete\testThird>                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                              [... cut ...]                                                                                                                                                                                                                                                                                                                                                                                                                                                           

Please review the issue and help me install the package.

song940, 2015-12-18:
sorry, this package is not supported your operating system right now .

closed 2015-12-18 by song940

kamaldeepoodles, 2015-12-18:
ok what are the requirements to run this project on windows OS i.e. which- windows OS, Visual Studio, Python, Node ?

song940, 2016-11-04:
@kamaldeepoodles just try again :)

mancioshell, 2021-06-10:
any update ? i can't run this pacakge on windows 10 and node 16