lsongdev/node-escpos #5: Libusb issue when initializing the library

cinoyter, 2016-05-21:
Hi!

I was getting this issue when running your sample program:

```
sami@jevich:~/Documents/projects/infodispenser$ node test_node_escpos.js 
/home/sami/node_modules/escpos/node_modules/usb/usb.js:168
	this.device.__claimInterface(this.id)
	            ^

Error: LIBUSB_ERROR_BUSY
    at Error (native)
    at Interface.claim (/home/sami/node_modules/escpos/node_modules/usb/usb.js:168:14)
    at /home/sami/node_modules/escpos/adapter/usb.js:22:11
    at Array.forEach (native)
    at USB.open (/home/sami/node_modules/escpos/adapter/usb.js:21:26)
    at new Printer (/home/sami/node_modules/escpos/printer.js:12:16)
    at Object.<anonymous> (/home/sami/Documents/projects/infodispenser/test_node_escpos.js:4:17)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)

```

After a little bit of research, I solved the problem by opening adapter/usb.js and moving the "iface.claim();" to after the try/catch. This seems to work for me! I'm not sure that this is the "right" way to solve the problem though.

Thanks for your great work!

jor3l, 2016-05-22:
Had the same issue and got it working the same way too. Also I have to run as sudo in order to be able to connect to the printer, probably something changed in the node-usb package

closed 2016-05-24 by song940

GuoJiKaiHehe, 2018-03-31:
I've had such a problem, too! Cry for help

Psychopoulet, 2018-04-03:
come from the USB package. I have to download the https://github.com/tessel/node-usb package and compile it separatly to work with

commit ef87979e (2016-05-24): fix #5