lsongdev/node-escpos #6: Error: Can not find printer

sadlyblue, 2016-06-11:
Hi,
I'm trying to print from node.js to a Epson Tm-T70II thermal printer connect with an usb cable.
I get the error "Error: Can not find printer" when trying the index.js. The printer is working. Even configured in cups, and i can print with another module (just not images/logos).
Can someone help, please?
Thank you

jor3l, 2016-08-22:
You will have to debug your USB devices, probably the printer is showing up with a different ID than the ones in the code causing the not found error

natsu90, 2016-09-22:
Hi, how do I debug this? The printer is working fine when in MacOSX but not in RPi.

jor3l, 2016-09-22:
Sudo works on my RPi

song940, 2016-11-01:
Did you solve the problem ? @natsu90

natsu90, 2016-11-01:
I just need to tighten up the usb, that's it.

song940, 2016-11-01:
[image]

@natsu90

closed 2016-11-01 by song940