lsongdev/node-escpos #9: How to use Network adapter?

pineapple507, 2016-08-19:
I changed a few lines of code, but I don't get it to work with an IP and Port? how i'm supposed to make it work?

jor3l, 2016-08-29:
Did you try opening a TCP connection to the wifi printer? then send the same commands, I was about to work on this but I don't have a WiFi printer to test, if you do, please let me know and maybe we can get this to work together

ArifADS, 2016-08-30:
I have not send the same commands, I have just send to it Text

jor3l, 2016-09-10:
@arifads do you have a wifi printer we can test?

ArifADS, 2016-09-10:
Yes I do, at the office, I have sent to it text and some raw commands like double height or cut. But is kinda pita to do it that way.

jor3l, 2016-09-10:
If you want we can try to get it working, I have Bluetooth printer and USB working so far

song940, 2016-11-01:
@pineapple507  Not implemented yet .

closed 2016-11-01 by song940

stefano-DM, 2017-05-16:
@jor3l how did you make the bluetooth printer work? thanks

zhaixiaoou, 2017-08-03:
@jor3l  how to connect the bluetooth printer?  Whether to use Serial Adapter? What is Serial Port? thanks!!!!

jor3l, 2017-08-03:
@stefano-DM @zhaixiaoou I got BT printer working for Cordova apps using the https://github.com/don/cordova-plugin-ble-central plugin. Basically I'm sending the same commands as this component does once I'm connected to the BT printer. Sorry for the late response.