lsongdev/node-escpos #13: 加油！如何打印图片呢？

keyiis, 2016-11-01:
感谢你的贡献！如果我想打印图片怎么办？

song940, 2016-11-01:
@keyiis see example https://github.com/song940/node-escpos/blob/master/examples/image.js

keyiis, 2016-11-01:
@song940 谢谢回复！下面这些 s8 d8等参数，分别代表什么，有哪些可选的，请指教，谢谢！
.image(image, 's8')
    .image(image, 'd8')
    .image(image, 's24')
    .image(image, 'd24')
    
    .raster(image)
    .raster(image, 'dw')
    .raster(image, 'dh')
    .raster(image, 'dwdh')

song940, 2016-11-01:
The second parameter is the`density` of [#image](https://github.com/song940/node-escpos/blob/master/printer.js#L269) method, 
the `s` means `single` and the `d` means `double`  in density,
the number `8` means image sampling rate is `8bit` and `24` also similar.

keyiis, 2016-11-01:
@song940  thank you very much!

closed 2016-11-01 by song940

keyiis, 2016-11-01:
@song940 使用image打印的结果见下图，我传递的image是dataurl
[image]

song940, 2016-11-01:
@keyiis  what `data-uri` do you use ?

keyiis, 2016-11-01:
@song940 是data-uri,用路径我也试过，printer.barcode 也试过，打印出的都是乱码，但是纯文本没有问题，我感觉问题应该在图片转buffer时造成的，正在看你的代码。不知怎么编译serialport特别慢，编译完后我再回复你node examples/image.js的结果。

keyiis, 2016-11-01:
@song940 
```
events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: LIBUSB_ERROR_NOT_FOUND
    at Error (native)
    at Device.usb.Device.open (D:\work\nodejs\node-escpos\node_modules\usb\usb.js:33:7)
    at USB.open (D:\work\nodejs\node-escpos\adapter\usb.js:77:15)
    at D:\work\nodejs\node-escpos\examples\image.js:9:10
    at D:\work\nodejs\node-escpos\image.js:51:5
    at D:\work\nodejs\node-escpos\node_modules\get-pixels\node-pixels.js:22:5
    at null.<anonymous> (D:\work\nodejs\node-escpos\node_modules\get-pixels\node_modules\pngjs\lib\png.js:74:7)
    at g (events.js:260:16)
    at emitOne (events.js:77:13)
    at emit (events.js:169:7)
```
打印机肯定处于连接状态。

dearvae, 2018-07-04:
git clone git@github.com:song940/node-escpos.git escpos
cd $_
npm i
node examples/image.js
大佬好，为什么我run你的file打印机没有反应呢，就一直在那不动，我想要打印图片，已经试过了可以打印barcode和文字了QAQ

dearvae, 2018-07-04:
哦不，他报错了这个QAQ
node examples/image.js
events.js:183
      throw er; // Unhandled 'error' event
      ^

Error: LIBUSB_ERROR_ACCESS
    at Device.usb.Device.open (C:\Users\BN\Dropbox\SchoolDocument\Year3semester1\FYP\node-escpos-master\node-escpos-master\node_modules\usb\usb.js:33:7)
    at USB.open (C:\Users\BN\Dropbox\SchoolDocument\Year3semester1\FYP\node-escpos-master\node-escpos-master\adapter\usb.js:100:15)
    at C:\Users\BN\Dropbox\SchoolDocument\Year3semester1\FYP\node-escpos-master\node-escpos-master\examples\image.js:9:10
    at C:\Users\BN\Dropbox\SchoolDocument\Year3semester1\FYP\node-escpos-master\node-escpos-master\image.js:51:5
    at C:\Users\BN\Dropbox\SchoolDocument\Year3semester1\FYP\node-escpos-master\node-escpos-master\node_modules\get-pixels\node-pixels.js:22:5
    at exports.PNG.<anonymous> (C:\Users\BN\Dropbox\SchoolDocument\Year3semester1\FYP\node-escpos-master\node-escpos-master\node_modules\pngjs\lib\png.js:74:7)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at exports.PNG.emit (events.js:211:7)
    at exports.PNG.<anonymous> (C:\Users\BN\Dropbox\SchoolDocument\Year3semester1\FYP\node-escpos-master\node-escpos-master\node_modules\pngjs\lib\png.js:36:10)

song940, 2018-07-05:
On Windows, Use Zadig to install the WinUSB driver for your USB device.

yhauxell, 2019-01-18:
Hi guys, @BeiNing:

I started having the same issue after updating my module and I found that
the current version of node-escpos has .image() function as async, because
of that you are getting a promise instead of printer object, so please
refactor accordingly wherever you use the method.

Hope this helps.

Yausell.

yhauxell, 2019-01-18:
Sorry, I just reply the wrong issue, that correct you are missing driver
:). But also you may have the same issue I described since
example/images.js is not up to date with the latest function signature.

Yausell.