lsongdev/node-escpos #14: LIBUSB_ERROR_NOT_FOUND

keyiis, 2016-11-02:
系统环境 win10 x64  nodejs 4.6
npm i 过程没有报错，但是运行下面的命令就报错
```
D:\work\nodejs\node-escpos>node examples/image.js
events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: LIBUSB_ERROR_NOT_FOUND
    at Error (native)
    at Device.usb.Device.open (D:\work\nodejs\node-escpos\node_modules\usb\usb.js:33:7)
    at USB.open (D:\work\nodejs\node-escpos\adapter\usb.js:77:15)
    at D:\work\nodejs\node-escpos\examples\image.js:9:10
    at D:\work\nodejs\node-escpos\image.js:51:5
    at D:\work\nodejs\node-escpos\node_modules\get-pixels\node-pixels.js:22:5
    at null.<anonymous> (D:\work\nodejs\node-escpos\node_modules\get-pixels\node_modules\pngjs\lib\png.js:74:7)
    at g (events.js:260:16)
    at emitOne (events.js:77:13)
    at emit (events.js:169:7)

```

song940, 2016-11-03:
duplicate to #12

closed 2016-11-03 by song940

apemen, 2020-04-21:
你好，问下，我用

> 

 ` 
const escpos = require('escpos');
escpos.USB = require('escpos-usb');
const device = new escpos.USB(0x1ad4,0x00c8);`
为什么么有反应，也没有报错也没有输出