lsongdev/node-escpos #27: Help with escpos

song940, 2017-02-28:
Hello,
 
I am using your node escape pos driver to use with a Star termal printer. I have a problem with USB code that I can’t understand. First time I open the device, it works but no more works in next printing.
 
This is my code:

```js
var escpos = require('escpos');
var device  = new escpos.USB();
var printer = new escpos.Printer(device);

app.post('/api/printer', function(req, res) {   
  var data = req.body;
  try{
    escpos.Image.load(data.img, function(image){          
      device.open(function(){
        printer
        .align('ct')
        .image(image, 's8')
        .cut(false, 3)
        .close(function(){
        });                                       
      });             
      res.json();
    });
  }catch(e){
    console.log("device.open error: ", e);
    res.json();
  }   
});
```
 
First time `self.endpoint = null` so everything is ok but next call, `self.endpoint` is filled so never call again callback function. What’s the trick? I tried closing device but is the same behaviour.
 
Can you help me please? What’s wrong?
 
Thanks in advance.
Víctor
Enviado desde Correo para Windows 10

song940, 2017-02-28:
```js
const escpos = require('escpos');

const device  = new escpos.USB();
const printer = new escpos.Printer(device);

app.post('/api/printer', function(req, res) {   
  var data = req.body;
  escpos.Image.load(data.img, printImage);
  function printImage(image){
  	printer
		.align('ct')
		.image(image, 's8')
		.cut(false, 3);
		res.json();
  }
});

process.on('exit', function(){
	device.close(function(err){
		console.log('goodbye!');
	});
});

device.open(function(err){
	if(err) return console.error("device.open error: ", e);
	app.listen(3000);
});
```

jor3l, 2017-02-28:
You have to open the device once, the code you posted tries to open it with each request. song940 version should fix your problem.

closed 2017-03-01 by song940