lsongdev/node-escpos #28: BASE64 image data url support?

rehmanShafi, 2017-02-28:
I am trying to use base64 image data url but am getting error
`TypeError: url.indexOf is not a function`

Is there any way around or am doing something wrong here?
`let device = new Escpos.USB();`
`let printer = new Escpos.Printer(device);`
`Escpos.Image.load(dataUrl, function(image) {`
   `device.open(function() {`
          `printer`
           `.align('ct')`
           `.raster(image)`
           `.cut()`
           `.close();`
         `});`
`});`

song940, 2017-02-28:
`url` is the path to the file. It can be a relative path, an http url, a data url, or an in-memory Buffer.
`TypeError: url.indexOf is not a function` means your url is not a `string` . Is your `url` correct?

rehmanShafi, 2017-02-28:
i am using Data Url `data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAaoAAAJFCAYAAACMdbM8AAAg.......`

rehmanShafi, 2017-02-28:
Solved it, data passed to `Escpos.Image.load` wasn't in perfect shape, a little bit cleaning got the job done.
Thanks for response.

vinothmoorthkumar, 2018-07-22:
How did you fixed this ??

Psychopoulet, 2018-07-23:
"data passed to Escpos.Image.load wasn't in perfect shape, a little bit cleaning got the job done"

ray-man, 2018-09-30:
The problem is back, Here are details, I am using BASE64 image data, 
`Escpos.Image.load(data.image,function(image) {
        device.open(function() {
          printer
            .align('ct')
            .image(image)
            .cut()
            .close();
        });
      });`
That's how my code look's like , result with `.image(image)` is as below
https://ibb.co/kUvsrz

Any suggestions ?

closed 2019-03-04 by song940