lsongdev/node-escpos #34: text height not working

pcapanna, 2017-03-29:
When setting other sizes other than 1x1, 1x2, 2x1 or 2x2, text height is always 1, and width wathever you put on it.

I have a not so happy solution, but solution at the very least:

I changed/added on commands.js:
```
_.TEXT_FORMAT = {
  ...
  TXT_PRE_SIZE    : '\x1d\x21', // other sizes
  TXT_OTHER_SIZE  : function(width, height){
    var hexWidth = width - 1;
    hexWidth = hexWidth.toString(16);
    var hexHeight = height - 1;
    hexHeight = hexHeight.toString(16);
    return hexWidth.concat(hexHeight);
  }
```
And on printer.js on size function:

```
Printer.prototype.size = function(width, height) {

  if (2 >= width && 2 >= height) {
    ...
  }
  else {
    this.buffer.write(_.TEXT_FORMAT.TXT_PRE_SIZE);
    this.buffer.write(_.TEXT_FORMAT.TXT_OTHER_SIZE(width,height), 'hex');
    // this.buffer.write(_.TEXT_FORMAT.TXT_HEIGHT[(8 >= height) ? height : 8]);
  }
```
Couldn't avoid using 'hex' to be the second parameter at write function.. maybe someone else could rewrite the function to return already escaped hex numbers.. Perhaps even concluding the change of size in only one write function wich would be cleaner.

Sophtware, 2017-04-09:
I ran into the same issue as the one you described. Thanks for putting up a potential solution. I'll give it a try if I find that I need it. :)

closed 2017-05-01 by song940

pcapanna, 2017-05-02:
Why was this issue closed? I can't see the changes in b6ac6b4. I will make a pull request about this so it's easier.

pcapanna, 2017-05-02:
#43

commit b6ac6b42 (2017-05-01): modofy indent and fix #34 , #36 .