lsongdev/node-escpos #35: USB Adapter, cannot close / disconnecting adapter

adenvt, 2017-04-08:
My `index.js`
```js
const escpos = require('escpos');

const device  = new escpos.USB();
const printer = new escpos.Printer(device);

device.open(() => {
  printer
    .text('Hello World')
    .feed()
    .cut()
    .close(() => {
      console.log('Close')
    })
});
```
I run via terminal
```bash
node index.js
```
Everything is OK... just script not stop, need to press <kbd>Ctrl + c</kbd> to stop it,
and close's callback never called.

I try change using Serial. work properly.

Note:
- OS: Ubuntu 16.04
- Node: 6.10.2

song940, 2017-04-10:
When u press <kbd>Ctrl + C</kbd>, terminal will send `SIGINT` to our program .
i guess `SerialPort` listen this signal and close the port on program exit, but usb does not do it.

so, we can add this code to our program ... 
```js
process.on('SIGINT', () =>   printer.close());

closed 2017-04-21 by song940

nadbm, 2018-02-18:
@song940  I think you understood the problem wrong. It is not that CTRL+C doesn't stop the code, but he is FORCED to kill the application. I will submit a PR to fix this shortly.

PR #108 (2018-02-18): Bugfix: Remove the detach listener once the device is closed, fix #35
Fix #35 

## Reproducing bug: 
test.js
```
var escpos = require('escpos')
var device = escpos.USB();
device.close()
```
```
node test.js
```

This hangs on the terminal, needs to be terminated
commit 82fca7d2 (2018-02-18): Remove the detach listener once the device is closed, fix #35
commit e0327416 (2018-02-18): Merge pull request #108 from nadbm/master

Bugfix: Remove the detach listener once the device is closed, fix #35