lsongdev/node-escpos #37: Example not working - truncated image qrcode

vekexasia, 2017-04-14:
Hello,

thank you for the project.

I've tried the following snippet with both USB and Serial device connection
[image]

the result is a truncated qrcode. (Every time at the same spot) .... I'm not sure if that's a bug in raster process or just a communication issue somewhere.
[image]

vekexasia, 2017-04-14:
After further investication:

Assuming that: the problem does not reside on my printer (which i don't think since other apps could print images just fine);

The problem might be in `image.js`.`toRaster()` i see width is getting "/8" but height is not. While the code is obscure to me it looked strange that just one dimension was going to be shrinked in the process.

I also saw that GS_V0 command is marked as obsolete and EPSON recommends using GS L (112 &50) to perform the same task. their documentation is too hard for me since I got in touch with esc pos standard just few hours back and I lack the needed knowledge to fully comprehend those parameters.

jor3l, 2017-04-14:
What about printing an QR code image? Does it get truncated too?

vekexasia, 2017-04-15:
With .Image?

vekexasia, 2017-04-15:
I used your image example with the following code and tux.png:
[image]

The result is the following:

[image]

Do images need to have a maximum size to be printed ? I tried with a "large" jpg but i got only black patterns printed.

uyu423, 2017-08-12:
I can not print the qr code normally either. I think there is a slight difference in specifications between printer manufacturers. In my case, the hardware does not seem to have a value defined for the qr code output. (I use an A7 printer) I think it is necessary to discuss how to solve the problem when specifications vary from manufacturer to manufacturer.

song940, 2017-08-13:
@vekexasia you can contact your printer's manufacturer, any questions u can reopen this issues .

closed 2017-08-13 by song940