lsongdev/node-escpos #38: sorry but i can´t do that

netico14, 2017-04-15:
I'm sorry it's my fault, I'm new to this, I ran the zadig program, select the printer, wait for it to finish and close the program, run the server.js and tell me that it can not find the printer, look for this problem but without success , Any idea of my error, thank you very much.

jor3l, 2017-04-15:
Share your code

netico14, 2017-04-15:
Hello jor3l, my code:
`const escpos = require('escpos');

const device  = new escpos.USB();
const printer = new escpos.Printer(device);

device.open(() => {
  printer
    .text('Hello World')
    .feed()
    .cut()
    .close(() => {
      console.log('Close')
    })
});`
and the error:
[url=http://subefotos.com/ver/?98ede98456f00ea35a9b2bda664ffc9bo.jpg][img]http://thumbs.subefotos.com/98ede98456f00ea35a9b2bda664ffc9bo.jpg[/img][/url]

thanks

netico14, 2017-04-15:
[image]

jor3l, 2017-04-16:
Seems to me that your PC is not configured well, are you able to print using another software?

netico14, 2017-04-16:
Thanks you, yes sir, i'm try from notepad

netico14, 2017-04-17:
Thanks, i can print now, between usb() need put the VendorID and ProductID, example usb(0x2B5, 0x204)

closed 2017-04-17 by netico14