lsongdev/node-escpos #39: Setting barcode height and width not working

pcapanna, 2017-04-24:
Setting barcode height and width is not working.
In fact the code isn't doing anything with width and height parameters other than checking them for then to use a default some default.

I made a working solution.

Changed in printer.js barcode prototype function:
```
Printer.prototype.barcode = function(code, type, width, height, position, font){
  if(width >= 2 || width <= 6){
    this.buffer.write(_.BARCODE_FORMAT.BARCODE_WIDTH[width]);
  }else{
    this.buffer.write(_.BARCODE_FORMAT.BARCODE_WIDTH_DEFAULT);
  }
  if(height >=1  || height <= 255){
    this.buffer.write(_.BARCODE_FORMAT.BARCODE_HEIGHT(height));
  }else{
    this.buffer.write(_.BARCODE_FORMAT.BARCODE_HEIGHT_DEFAULT);
  }
```

and changed some things in commands.js, all barcode size related stuff. Had to add a function for barcode_height in fact to not have a large 255 size array:

```
  BARCODE_HEIGHT  : function(height){
    return '\x1d\x68'+String.fromCharCode(height);
  },
  BARCODE_HEIGHT_DEFAULT: '\x1d\x77\x64',

  BARCODE_WIDTH  : { // Barcode Width  [1-5]
      1: '\x1d\x77\x02',
      2: '\x1d\x77\x03',
      3: '\x1d\x77\x04',
      4: '\x1d\x77\x05',
      5: '\x1d\x77\x06'
  } ,
  BARCODE_WIDTH_DEFAULT: '\x1d\x77\x03',
```

Acording to ESC/POS documentation you can set height from 1 to 255. But don't know why my ESC/POS printer only works with values from like 10 to 128 (which is more than enough for me), other values I get printed 'UNDEFINED'. It could be that my printer does not support all sizes or that I'm not understanding or doing something correctly. It is something to be aware.. It would be great if someone else could test this.

jor3l, 2017-04-24:
Hi pcapanna, can you send a Pull request? It will be easier to test and integrate into the code for song.

pcapanna, 2017-04-24:
Didn't know how pull requested worked. I just made one: #40

closed 2017-04-24 by pcapanna