lsongdev/node-escpos #41: LIBUSB_ERROR_ACCESS

altabz, 2017-04-25:
I am new to Linux, and im not sure how to solve a problem.
:
Error: LIBUSB_ERROR_ACCESS
    at Error (native)
    at Device.usb.Device.open (/home/alejandro/node_modules/usb/usb.js:33:7)
    at USB.open (/home/alejandro/node_modules/escpos/adapter/usb.js:81:15)
    at Object.<anonymous> (/home/alejandro/Dropbox/scripts/nodejs/escpos.js:9:8)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)

I would like to know why I have this problem, I use UBUNTU and I followed the instruction "sudo apt-get install build-essential libudev" that apears in the documantation.

Can you help me resolve this problem? And perhaps show me step by step. 
Thanks*

song940, 2017-04-25:
sudo ...

altabz, 2017-04-25:
Hi!
This is my next error :(

alejandro@pc:~/Dropbox/scripts/python$ sudo python escpos.py
Traceback (most recent call last):
  File "escpos.py", line 3, in <module>
    p.text("Hello World\n")
  File "/home/alejandro/.local/lib/python2.7/site-packages/escposprinter/escpos.py", line 285, in text
    self._raw(text)
  File "/home/alejandro/.local/lib/python2.7/site-packages/escposprinter/printer.py", line 61, in _raw
    self.device.write(self.out_ep, msg, self.interface)
  File "/home/alejandro/.local/lib/python2.7/site-packages/usb/core.py", line 940, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/home/alejandro/.local/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/alejandro/.local/lib/python2.7/site-packages/usb/core.py", line 215, in setup_request
    intf, ep = self.get_interface_and_endpoint(device, endpoint_address)
  File "/home/alejandro/.local/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/alejandro/.local/lib/python2.7/site-packages/usb/core.py", line 231, in get_interface_and_endpoint
    raise ValueError('Invalid endpoint address ' + hex(endpoint_address))
ValueError: Invalid endpoint address 0x1

lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 04f3:0363 Elan Microelectronics Corp. 
Bus 003 Device 011: ID 0483:070b STMicroelectronics  <------------------------------------------ this is i need
Bus 003 Device 003: ID 04f2:b446 Chicony Electronics Co., Ltd 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

thanks!!!! for help me

song940, 2017-04-26:
python? node-escpos is a JavaScript package.

altabz, 2017-04-26:
sorry wrong post! jeje

closed 2017-05-01 by song940

uyu423, 2017-08-03:
I think it should be available even if not the root user. Is there a way?

wanxsb, 2017-09-18:
I also have  this problem, sudo not working. BTW, I use mac

zakaria-chahboun, 2021-07-18:
### Linux Print without *sudo permission*
Add a new file in:
- `/etc/udev/rules.d` named `usb.rules`

It should have the following contents:
- `SUBSYSTEM=="usb", MODE="0666"`

You can do this in terminal quickly: `sudo nano /etc/udev/rules.d/usb.rules`

Reload rules:
- `sudo udevadm control --reload`

> This will make all USB devices readable and writable by all users.