lsongdev/node-escpos #42: Can not find printer

himanshu21092, 2017-05-02:
Hi, I am using this in my electron app. Name of the printer is Epson TM-T20 and I have connected using a usb. Module is not able to find my printer where as test printer button is working in my printer settings.

Here is my code :-

const escpos = require('escpos');
const device  = new escpos.USB();
console.log(device.getDevices());

jor3l, 2017-05-02:
Hi, the test button has nothing to do with the PC the printer is connected to. Try to print using another software in order to make sure you have all the drivers needed for the printer to work, then try with this plugin again. Cheers

himanshu21092, 2017-05-02:
Hi @jor3l, I was talking about the test button on my linux settings not on the printer. Driver is installed and I am able to take print using other modules. But I would like to use this one. I am attaching the screenshot of the settings page.
[image]

song940, 2017-05-02:
@himanshu21092 `USBAdapter` does't have `usb#getDevices` method, it's `usb#findPrinter`.

and, if u see error as `Can not find printer`, u can run

```bash
# lsusb
Bus 002 Device 001: ID 04b8:0202 Epson ...
```
put vid and pid to usb adapter's constructor 
 
```js
const device = new escpos.USB(0x04b8, 0x0202);
const printer = new escpos.Printer(device);
```

command

himanshu21092, 2017-05-03:
@song940 , I did that. Now it is not throwing exception there but when I am trying to run 

device.open(function(){

  printer
  .font('a')
  .align('ct')
  .style('bu')
  .size(1, 1)
  .text('The quick brown fox jumps over the lazy dog')
  .text('敏捷的棕色狐狸跳过懒狗')
  .barcode('12345678', 'EAN8')
  .qrimage('https://github.com/song940/node-escpos', function(err){
    this.cut();
    this.close();
  });

I am getting this exception : Error: LIBUSB_ERROR_ACCESS
    at Device.usb.Device.open

I have one more question, is it possible that I can discover the printers already installed whether it is wireless or usb and send print commands?

song940, 2017-05-19:
`sudo ...`

closed 2017-05-19 by song940