lsongdev/node-escpos #44: Print accent (é - è - ù - ...)

sanytol, 2017-05-04:
Hi, currently working on a NodeJS app. I would like to print accent, but I'm not able to made it work, only printing some weirds symbols. I do try with different encoding (UTF-8, Windows-1250, ...), but same issue.

Here is my code :
```
// Include lib
var escpos = require('escpos');

// Define device and printer
const device = new escpos.Serial('COM1');
const printer = new escpos.Printer(device);

// Open connection
device.open(function(){
     printer
      .font('a)
      .align('lt')
      .style('normal')
      .text('Héllô Wùrldè', 'UTF-8');
});
```

According to comments in `printer.js`, method `text` can only print alpha-numeric value, but method `print` and `println` do not have\use `encoding` parameter.

Any help would be appreciate !

emrealparslan93, 2017-05-23:
@sanytol @ArifADS @aroms 

Try to find encoding from this table : https://github.com/SheetJS/js-codepage#generated-codepages

**Example:** If you're going to use Turkish characters ('ÖöÜüĞğŞşİıÇç'), you need to find this row:
`857	unicode.org	OEM Turkish; Turkish (DOS)`
And use `857` for encoding.

```
// Open connection
device.open(function(){
     printer
      .font('a)
      .align('lt')
      .style('normal')
      .text('ÖöÜüĞğŞşİıÇç', '857');
});
```

aroms, 2017-05-24:
It doesn't work. tested under ```Excelvan HOP-E801```.

emrealparslan93, 2017-05-24:
@aroms Did you set a proper '`Code Page`' to your `Excelvan HOP-E801` ?

aroms, 2017-05-25:
@emrealparslan93 Yes the ```Code Page``` is ```OEM857(Turkey)``` and it doesn't work. Prints strange characters as ```℠S¬½ ^™€``` when it would have to be ```ÖöÜüŞğŞşİıçç```

emrealparslan93, 2017-05-25:
@aroms It's strange. It works on my thermal printer `Possify FY830x`.

aroms, 2017-05-25:
Current Code Page: 61 (61 = OEM857 (Turkey))
[image] 
[image]

Do you know why it does not work? what is wrong?

emrealparslan93, 2017-05-25:
@aroms 
In my case, firstly I set `Default Code Page` of my printer as `61`. Then I `restarted` thermal printer. After that, I set encoding as `857` in text() function. Here is my function:

          const printer = new escpos.Printer(device);
            device.open(function(err, device) {
                if (err) return reject(err);
                try {
                    printer
                        .font('a');
                        .align("lt")
                        .size(1, 1)
                        .text(text,'857')
                        .println();

                    printer.control("LF")
                         .cut()
                         .close();
                    return resolve(true);
                } catch (err) {
                    return reject(err);
                }
            });

If these steps does work in your case, I recommend you to talk with your thermal printer's manufacturer. They can help I guess.

uyu423, 2017-08-12:
Has this issue been resolved yet? I think it would be helpful to refer to the iconv-lite documentation. The Printer.text method encodes the buffer using iconv-lite. I use the CP949 encoding well by referring to this document.

https://github.com/ashtuchkin/iconv-lite/wiki/Supported-Encodings

song940, 2017-08-13:
@sanytol you can try @uyu423 's solution, reopen issues if u have any questions  .

closed 2017-08-13 by song940

Awoogamuffin, 2019-04-06:
I know this is quite an old thread, but I'm having the same issues. The character set I need is Spanish, along with the Euro symbol. Following the links on this page, I gather the encoding I need is '858' (OEM Multilingual Latin 1 + Euro symbol) and though it does indeed print the Spanish characters I need (é á í ñ etc.) it does not print the euro symbol. Any idea what I need to do?

3h3c, 2019-12-02:
@Awoogamuffin When you have a (Excelvan, HOI) HOP-E801 you have to set firstly the codepage by the printer tool. In the code set the Encoding CP858

`var device = new escpos.Network('192.168.x.x')
var printer = new escpos.Printer(device, {encoding: 'ISO-8859-1'})`

now send firstly a special buffer to cancel Kanji character mode

`device.open(() => {
    printer.print(new Buffer([28, 46])); //<<<<<cancel Kanji character mode
    printer.font('A');
    printer.align('LT');
    printer.size(1, 1);
    printer.style('NORMAL');
    printer.text(''ÄÖÜß€);
    printer.feed(1);
    printer.cut();
    printer.close();
})`

hope it will help you

terehov, 2021-03-24:
@3h3c Thank you so much for your hint with the Kanji character mode!