lsongdev/node-escpos #45: Can we send an HTML or PDF file to print?

burakkilic, 2017-05-05:
Hello,

We need to print aligned rich text for restaurant receipts.

[image]

What is the best way to to that?

Psychopoulet, 2017-05-11:
add some spaces. I created a wrapper for a project XML > ESCPOS which automaticly add some spaces to the ticket for that, you can do the same...

closed 2017-06-19 by song940

conquera99, 2020-09-14:
hi, do you have an example for this?

currently the print output is not consistent in different printer

thank you

conquera99, 2020-09-14:
> hi, do you have an example for this?
> 
> currently the print output is not consistent in different printer
> 
> thank you

or did you mean to send XML to `.text()` method?