lsongdev/node-escpos #48: Barcode issues (again)

adamxp12, 2017-05-18:
So EAN-8 barcodes print great but they are very limited vs something like UPC-A or CODE128.

Using an EPSON TM-T88III which according to the technical reference guide ([http://www.i-o.cz/user/upload/TM88%20III/l7O-TM-T88II_TechnicalRefGuide.pdf](http://www.i-o.cz/user/upload/TM88%20III/l7O-TM-T88II_TechnicalRefGuide.pdf)) supports UPC-A, CODE128 and the other types supported by this library.

But all I get is blank spaces where the barcode should go (like the printer is seeing a barcode is trying to be printed but fails to generate it) when I try to print anything other than EAN-8

I doubt it is a hardware issue as judging by another issue someone had with an EPSON TM-T20 only printing EAN-8 barcodes it must just be some incompatibility with EPSON printers.

Any one have any ideas for a fix. generating images is not possible simply because they take ages to print and is harder to code for (me like simple)

song940, 2017-09-16:
How about now? Any questions reopen this issues.

closed 2017-09-16 by song940

yelhouti, 2021-05-03:
for CODE128 I found out that you need a leading {B in the message sent, still figuring out how to that, for anyone else facing the same issue