lsongdev/node-escpos #49: How to return error if ethernet printer is not found?

burakkilic, 2017-05-19:
I'm trying this code but it doesn't work:
   
      device.open(function(err, device) {
                if(err) reject(err);
     });

How to handle connection errors?

song940, 2017-05-27:
for example: 

```js
'use strict';
const escpos = require('escpos');

const device  = new escpos.Network('never-exists-remote-printer');

const printer = new escpos.Printer(device);

device.open(function(err){

  return console.error(err);

});

```

will output

```bash
➜  node-escpos git:(master) ✗ node examples/index.js
{ Error: getaddrinfo ENOTFOUND never-exists-remote-printer never-exists-remote-printer:9100
    at errnoException (dns.js:28:10)
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:73:26)
  code: 'ENOTFOUND',
  errno: 'ENOTFOUND',
  syscall: 'getaddrinfo',
  hostname: 'never-exists-remote-printer',
  host: 'never-exists-remote-printer',
  port: 9100 }

```

closed 2017-06-19 by song940

imaadhrizni, 2018-10-19:
Hi,
This exact code now throws this:
```
{ Error: connect ETIMEDOUT 10.48.224.47:9100
    at Object._errnoException (util.js:992:11)
    at _exceptionWithHostPort (util.js:1014:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1186:14)
  code: 'ETIMEDOUT',
  errno: 'ETIMEDOUT',
  syscall: 'connect',
  address: '10.48.224.47',
  port: 9100 }
```

I am unable to get a response for about 30 seconds after which this above error is thrown.

Psychopoulet, 2018-10-19:
it's because you cannot reach the printer (with this IP & this port) from this computer

imaadhrizni, 2018-10-19:
@Psychopoulet how come it throws me an `ETIMEDOUT` instead of an `ENOTFOUND` is my concern. I am currently unable to check if a printer was reachable at a given IP & Port since it times out rather than throwing up an exception for not being found. I'm just wondering if this is the expected outcome from the code provided by @song940 above

Psychopoulet, 2018-10-22:
is it the right network ? I mean, for exemple, don't you try to connect a printer which is on an ethernet network (with specifics IPs), but being in WIFI (with other specifics IPs) ?

imaadhrizni, 2018-10-24:
No. For instance taking a Wifi Network with a gateway of 192.168.1.1 and an ethernet printer that is available at 192.168.0.24. If perchance the printer is turned off then I would expect the code to throw me a ENOTFOUND as per @song940 answer above. I was just wondering if this is the expected behavior as sometimes it takes 30 seconds + to throw the timeout error when trying to connect to a network printer that is turned off.

P.S. I am basically trying to come up with a small status to be thrown to the user as a form of feedback if the printer is not found and am currently using a `Promise race` to do it. However since I found this issue it seems a more elegant way of handling it since the `ENOTFOUND` is available

aroms, 2018-12-18:
In my system the default timeout is 1 minute and 15 seconds, more than a minute is too much for a timeout of this type, as much, I think it would have to be 10 seconds... How can reduce this timeout? (Basically, getting ETIMEDOUT faster) Any ideas?

jor3l, 2019-02-13:
Use [https://nodejs.org/api/net.html#net_socket_settimeout_timeout_callback](https://nodejs.org/api/net.html#net_socket_settimeout_timeout_callback) - `this.device.setTimeout(1000)`

nameszian, 2019-02-14:
@jor3l It shows that you have no idea what you are talking about, that only serves to add more time, you should read everything before commenting...

jor3l, 2019-02-14:
Please explain how setting a timeout adds more time to it

nameszian, 2019-02-14:
To not repeat it, read it here: https://github.com/song940/node-escpos/issues/181#issuecomment-463604971

joseluisrm, 2019-05-24:
Hi, this worked for me, I hope it's useful:

   ```
connectPrinter(address:string):Promise<any>{
   return new Promise<any>(
      (resolve, reject) => {
    const connection = new escpos.Network(address);
    const printer = new escpos.Printer(connection);
    connection.open((err) => {
            if (err) {
              console.error(err);
              reject(err);
            }
      });

    connection.device.setTimeout(TIMEOUT_PRINTER, () => {
          console.log("socket timeout");
          connection.device.destroy();
          let message = "Timeout connection - address:" +address;
          let err = new Error(message);
          console.error(err);
          reject(err);
    });

    connection.device.once('connect', () => {
          console.log("socket connect");
          connection.device.setTimeout(0);
          //Ready to use printer or return 
         resolve(printer)
    });
});
}
```

regards!