lsongdev/node-escpos #52: Printing issue

toure20, 2017-07-04:
Could you describe how to align text to the left and right in one line?
[image]

song940, 2017-07-04:
You can align left, then print, and then align right print right .

like this: 

```js
printer
.align('left')
.print('SM Water')
.align('right')
.print('0.00')
.flush();
```

toure20, 2017-07-04:
Thank you for reply! I tried by that way, but after setting second align, text prints at new line. Is it depends on pos-printer type?

toure20, 2017-07-04:
Thanks for your quick reply:) I tried by that way, but text (i mean 0.00) after new alignment printed in new line. May be it depends on what kind of pos-printer i have?
[image]

Psychopoulet, 2017-07-05:
you can pad it with spaces

closed 2017-07-13 by song940