lsongdev/node-escpos #53: Is this printer compatible with node-escpos?

emrealparslan93, 2017-07-21:
Hello, 
I'm using Mac and I'm trying to use this printer https://epson.com/For-Work/Point-of-Sale/POS-Printers/TM-T88V-POS-Receipt-Printer/p/C31CA85084 with this library. But it doesn't see the printer although I installed the driver of it. Is it possible that this printer does not support ESC/POS commands? Thanks.

Hayvanimal, 2017-07-21:
Probably that printer supports ESC/POS commands, but '**node-escpos**' is using '**libusb**' library...
Btw '**node-escpos**' has no reletion to specific printer drivers i think..

Try this..

`brew install libusb-compat `

İyi Çalışmalar..

Psychopoulet, 2017-07-24:
escpos modules (node, python, php, etc...) are essentialy based on the EPSON standards, which are used by 80% of the printers. It can append that a specific printer does only what it want, but with an EPSON it should work well.

emrealparslan93, 2017-07-24:
@Hayvanimal @Psychopoulet Thank you both for your answers. The issue was completely different. Sorry for inconvenience.

closed 2017-07-24 by emrealparslan93