lsongdev/node-escpos #60: Suggestions for JavaScript convention rules, including eslint

uyu423, 2017-08-13:
Thank you very much for letting me work together as a collaborator. But I will not push it directly into the master branch. All commits will be made through pull requests.

I think we should set coding rules for our escpos package. And I hope to have a serious discussion on this topic. In fact, I was very embarrassed when I first checked the source code for the escpos package. Because it was the first coding rule to be born and born.

I personally would like to follow the coding rules that are common to many developers around the world. For example, JavaScript coding conventions such as eslint-config-airbnb will be very powerful and effective rules. If we follow Airbnb 's coding rules, I think external developers will be a little easier to contribute to our open source.

We have drawn some coding styles that will change if we follow the rules of airbnb.

## printer.js (1~10 line)
[image]

## Printer.prototype.align
[image]

## Printer.prototype.size
[image]

## Printer.prototype.image
[image]

## Network.prototype.open
[image]

All of the above changes are the result of using eslint's auto-fix feature. Eslint shows me that there are 162 problems left after auto-fix.

[image]

I have committed the automatically modified source code to my private repository branch. Ask for confirmation. (https://github.com/uyu423/node-escpos/tree/feature/eslint-airbnb)

I have not tried the actual test because it is still a suggested step. We need to verify the compatibility of Airbnb's rules with each version of Node.js, and it's a matter to consider which version to support. If Airbnb's rules cause big problems with package compatibility, we can use Google or Standard JavaScript rules instead of Airbnb. Even if we can not use everything, we should at least specify a minimum rule like express.js. (https://github.com/expressjs/express/blob/master/.eslintrc)

I want to hear feedback.

uyu423, 2017-08-13:
The issue of changing the coding style and forcing it is a very cautious issue. Therefore, I would like to hear as many people as possible.

Psychopoulet, 2017-08-20:
I think everyone has his own personnal coding rules, so it should be difficult to force everyone to use only one's convention...
it can be interesting to add a precommit hook to force some common eslint rules, I guess... but to my mind, it's far more important to have a working, tested and complete package.

uyu423, 2017-08-21:
@Psychopoulet okay. I understand.
I've checked the source code of several open source packages since I created the issue, but I did not have much code to follow eslint.
I do not think I need to discuss this issue anymore.

closed 2017-08-21 by uyu423