lsongdev/node-escpos #62: Image printing issues

nicolasbadia, 2017-08-16:
Hi,
I need to print a canvas ticket. For this, I convert it to base64 (with toDataURL) and then use `escpos.Image.load` to convert it. I then print it using `raster`.
My issue is that `escpos.Image.load` can take a very long time (20 seconds if the ticket is a bit long).
Futhermore, the image is not very well printed.
Here is a picture where you can see on the left, the canvas printed using an ePos library (with the ePOS-Print API Version 5.0.0 of epson), and on the right the canvas printed using this library.

[image]

So I'm wondering if there is anything I can do to improves the speed and the quality of the print.
I can contribute to this library, but I don't really know where to start...

nicolasbadia, 2017-08-16:
Looking at the specifications of the printers, the TM-T70-i is 180 DPI while the TM-T20II is 203 DPI.
I guess the width difference comme from there. Do you know if there is a way to get the DPI of the printer with the library?

jor3l, 2017-08-16:
I'm surprised you were able to print it that good lol. Anyways, maybe some settings object should be defined because there are so many printers out there and as far as I could tell, there is no way to get that info from the printers I got, not even if the paper is missing.

nicolasbadia, 2017-08-16:
Thanks for the answer. 
Indeed, I agree that adding a setting on the printer definition to choose between 180 and 203 might be the best solution.

For my main issue which is a very long processing time, you have no idea how to improve it?

jor3l, 2017-08-16:
Multiple processes doing the image conversion at the same time should improve the speed but I haven't tried images yet

nicolasbadia, 2017-08-16:
I found C utility call png2pos to convert PNG images to ESC/POS.
It convert my base64 images instantly and print them perfectly (exept if I use the `-p` option. In this case, it print them like this library).
Here is an example for people having the same trouble:

    const fs = require('fs'),
      spawn = require( 'child_process' ).spawn,
      escpos = require('escpos'),

    var env = Object.create(process.env);
    env.PNG2POS_PRINTER_MAX_WIDTH = 576;

    var image = data.image.replace(/^data:image\/png;base64,/, "");
    fs.writeFileSync("/tmp/image.png", image, 'base64');

    var png2pos = spawn('png2pos', ['/tmp/image.png'], { env: env });
    png2pos.stdout.on( 'data', stdout => {
      var device = new escpos.Network('192.168.1.5'),
        printer = new escpos.Printer(device);

      device.open(function(err) {
        printer.buffer.write(stdout);
        printer.cut();
        printer.close();
      });
    });

closed 2017-08-16 by nicolasbadia

sam-lex, 2018-01-18:
Just fyi, png2pos doesn't support windows anymore.  
https://github.com/petrkutalek/png2pos/issues/15