lsongdev/node-escpos #63: Arabic Langugae

asaadbaw, 2017-08-17:
Hello

How can I print Arabic letters? Even the Chinese letters in the example not printing correctly!

uyu423, 2017-08-17:
I use Korean. If you want to print a specific language, you need to use an encoding type that supports Arabic as the second parameter in the Pinter.text method. See the following URL.
https://github.com/song940/node-escpos#texttext-encodetype

uyu423, 2017-08-17:
I think it will work fine if you code it like this:
```javascript
device.on(function() {
  printer.text(arabicString, '1256');
  printer.cut();
});
```

If you have any other problems, please give additional comments.

closed 2017-08-17 by uyu423

asaadbaw, 2017-08-21:
Hello
I managed to print Arabia using 
  .text('أسعد البو','ISO-8859-6')
but the printed text get reversed like this:
وبلا دعسأ
Any help with that?

uyu423, 2017-08-21:
@asaadbaw It is a very strange situation.
I have never used a character to write from right to left and I do not think it will help.
It will obviously be printed in the order in which they are entered into the buffer, should Arabic characters reverse the buffer input?
I think it would be helpful for you to check the following source code yourself.
https://github.com/song940/node-escpos/blob/master/printer.js#L81

asaadbaw, 2017-08-21:
umm! I'll check.
one more thing. Can I run this library from angularjs from the client side??

asaadbaw, 2017-08-21:
> @asaadbaw <https://github.com/asaadbaw> I know it is impossible. This is
> because you can not control system resources on the Web. (As far as I know,
> the Node.js native module is not available on the web.)
> You can also use a multi-platform native program development environment
> like Electron.js to solve the problem.
> I am using a printer in the Windows operating system using Electron.js

I am using cordova so the it will be running from either android or ios.
Any hint?

uyu423, 2017-08-21:
@assadbaw I have no idea how you want to use a POS printer on your mobile device.
Are you trying to connect a POS printer to your mobile device?

I know that phonegap and cordova only make web pages look like mobile native apps.
Assuming you connect a POS printer on a mobile device, the cordova project must be able to use the resources of the mobile device using the Node.js Native Module. I do not know if it is possible.
There seem to be developers who have tried the same thing with you. You should refer to this discussion or try searching with similar keywords.

https://stackoverflow.com/questions/14513029/phonegap-with-node-js