lsongdev/node-escpos #68: Just Blank paper - DYMO LabelWriter 450

lukasFrajt, 2017-08-25:
Hi, I tried example code. Connection to printer is okay, but I always got just blank paper. My printer is Dymo LabelWriter 450
 
My code
```
var express = require('express');
var router = express.Router();
const escpos = require('escpos');

const device  = new escpos.USB();
const printer = new escpos.Printer(device);

/* GET print. */
router.get('/print', function (req,res,next) {

    device.open(function(err){
        printer
            .font('a')
            .align('ct')
            .style('bu')
            .size(1, 1)
            .text('The quick brown fox jumps over the lazy dog')
            .text('敏捷的棕色狐狸跳过懒狗')
            .cut()
            .close()

    });
});

module.exports = router;
```

uyu423, 2017-08-25:
I think you should check out a few things.

1. What is the default encoding setting for the printer you are using?
2. The default character encoding setting for `escpos` is `GB18030`. (Chinese)
3. If the printer does not support `GB18030`, change the character encoding value in the `Printer.text()` method (see https://github.com/song940/node-escpos#texttext-encodetype)

uyu423, 2017-08-25:
I checked the following documents.
http://download.dymo.com/dymo/technical-data-sheets/LW%20450%20Series%20Technical%20Reference.pdf

I suggest you use the Printer.text () method as follows.

```javascript
device.open(function () {
  printer.text('The quick brown fox jumps over the lazy dog', 'ascii');
  printer.cut();
});
```

song940, 2017-09-16:
Please check compatibility of your printer.

closed 2017-09-16 by song940