lsongdev/node-escpos #71: Image Print Delay 7 Seconds

RUSHt, 2017-08-25:
I have an image (68k 512px x 860px).

escpos.Image.load() takes just over 7 seconds.

Is this to be expected, or am I doing something wrong?

(I have previously used the EPSON ePOS JS SDK, with the same images and the delay was imperceptible). 

Best Regards
Peter

song940, 2017-09-16:
Can you post your image ? And nodejs versions.

PeterA1, 2017-09-16:
The client environment is nw.js, which is using 8.5, on the same platform I have 6.11.1.  I can't recall if the lib ran a service in node, or was operating directly in nw.js with node 8.5.  I think it was a service since there were issues talking to serial devices from nw.js with your lib. Attached are sample images.

[image]
[image]

I have tried alternative approaches, and all seem deficient in some way. Epson have an SDK, but it won't talk to serially connected devices, another I've tried prints with ESC / * (a line at a time), and although this starts quickly the output itself is slow.  In all cases the image is sub-optimal (even with the Epson SDK).

myq1991, 2019-02-19:
Is any update of this issue? I meet same problem when I using OrangePi to control the pos-printer. The function "escpos.Image.load()" spend about 3 seconds.

artty58, 2019-09-06:
how to for data

Psychopoulet, 2019-09-06:
... what ?

tloriato, 2019-11-27:
Do we have any idea why it's taking so long? Is it possible for us to look into alternative libraries instead of get-pixels?

PeterA1, 2019-11-28:
FWIW I ended up changing out the customer printers to Star devices,  then used WebPRNT to send XML structured documents to the printer over WiFi / LAN ( I use TSP654 ).  The printer also supports a thing call cloudPRNT and with this you can de-spool from a server ( using the same WebPRINT XML documents ). It all a bit slow and awkwardly wrong, but it seems these people ( Epson / Star ) stopped developing their software in about 2007 and simply can't be bothered to do the obvious thing.