lsongdev/node-escpos #72: 打印出来乱码中文也没有支持！能告诉是什么问题么！

bacel12596, 2017-09-02:
`const escpos = require('../');
const device  = new escpos.Network('192.168.0.101');
const printer = new escpos.Printer(device);
console.log(printer);

device.open(function(err){

  printer
  .font('a')
  .align('ct')
  .style('bu')
  .size(1, 1)
  .text('The quick brown fox jumps over the lazy dog')
  .text('敏捷的棕色狐狸跳过懒狗')
  .barcode('1234567', 'EAN8')
  .qrimage('https://github.com/song940/node-escpos', function(err){
    this.cut();
    this.close();
  });
});`

[image]

bacel12596, 2017-09-02:
二维码也没有

bacel12596, 2017-09-02:
麻烦，可以告诉我么

uyu423, 2017-09-05:
I think you should check the specifications of the printer you are using. Does the printer support Chinese character encoding settings and 2D barcode printing?

Psychopoulet, 2017-09-12:
o_O
how do you think we can help you writing in a non-english way ? why not in cyrillic ? ^^

whatever, I agree with uyu, but you can test first with an encodage enforcement (encode("encodeType"))

song940, 2017-09-16:
Check encoding of your printer supports.

In chinese:

现在还有问题吗？检查下你的打印机编码。

closed 2017-09-16 by song940

bacel12596, 2017-09-16:
已经好了谢谢！ ok  thank you

song940, 2017-09-16:
How did you solved it ?

In chinese:

普通打印和二维码都没问题了？你怎么解决的？
你有修改代码吗？如果有欢迎 PR 给我。