lsongdev/node-escpos #74: How do I handle if usb printer is off or disconnected?

lutherpros, 2017-09-06:
Tried this does not work:
const escpos = require('escpos');
const device  = new escpos.USB();
const printer = new escpos.Printer(device);
device.open(function(err){

  return console.error(err);

});

jor3l, 2017-09-06:
Does the printer show up even when its off? (device found?)

song940, 2017-09-16:
You can listen disconnect event on device object.

closed 2017-09-16 by song940

sam-lex, 2017-12-30:
Hi, I couldn't find an example anywhere about how to handle disconnected, do we simply listen to that event and if it happens, do we just open it again? is there something else we need to do?

I'm trying to make a simple Print Server. (I don't have a printer with me so I can't test properly)

Could someone check the code below and see if that would work?  
In the examples I see `printer.close()`. Do we need to call that every time after we print? or should we use only `.cut()` and call `.close()` on server shutdown?

```typescript
import * as escpos from 'escpos';

class PrintManager {
  device: any;
  printer: any;
  isConnected = false;

  constructor() {
    this._init();
  }

  print(str: string) {
    if (!this._validateInitSuccess()) return; // Make sure is initialized
    if (!this._openDevice()) return; // Make sure device is connected
    this.printer
      .text( str )
      .cut()
      .close(); // Do we need close here?
  }

  // Returns true if initialized successfully
  private _init(): boolean {
    try {
      this.device = escpos.USB();
      this.printer = escpos.Printer( this.device );
      this.device.on('disconnect', () => { // Handle Disconnect
        this.isConnected = false;
      });
      return true;
    }
    catch (e) {
      console.warn('Could not initialize printer: ', e);
      return false;
    }
  }

  // Returns true if initialized correctly
  // Reinitializes if previous _init() failed
  private _validateInitSuccess(): boolean {
    const initialized = !!this.device && !!this.printer;
    return initialized || this._init();
  }

  // Returns true if device opened successfully
  private _openDevice(): boolean {
    if (this.isConnected) return true;
    try {
      this.device.open(() => this.isConnected = true );
      return true;
    }
    catch (e) {
      console.warn('Could not open device: ', e);
      return false;
    }
  }

}

const printManager = new PrintManager();

printManager.print('hello');
printManager.print('world');
```

trendy0413, 2023-04-03:
Can I open the device twice at the same time?
`let commands = new Uint8Array([...]);
const print = () => {
  devices[place].open((err) => {
          if(err) {
            reject({
              msg: `printer is not connected`
            })
          }
          devices[place].write(commands);
          resolve();
        })
}
print()
print() // call twice
`