lsongdev/node-escpos #76: React native ?

ax88, 2017-09-08:
Does it support react native ?

uyu423, 2017-09-09:
I am not react-native but I am using it well in electron. React-native also seems to be available if module rebuild works well.

ricardocjca, 2017-09-10:
I tried to do with electron, but to get lost in the zadig installation, I love the printer driver without the driver and driver, now the printer driver is a serial bus device, I do not know if I did it right, it's like me help I can not run the example.

uyu423, 2017-09-12:
@ricardocjca 
I was not good at using serial port communication. Nevertheless, if you want to use the serial port, make sure that the COM port setting is correct. Many printers may need to install additional driver files to configure the COM port. 

I used POSBANK's A7 printer and had to install the serial port driver. Please refer to the printer manufacturer's website you are using.

I think it's better to do USB communication and install zadig. When I used serial port communication it was very unstable.

song940, 2017-09-16:
You can expose printer to network and use network adapter in react native .

closed 2017-09-16 by song940

brkrtp, 2019-06-10:
+1 ı want to using react native esc pos only network module, but ı dont have idea how its work...