lsongdev/node-escpos #80: Module not found: Error: Can't resolve 'net', 'fs', 'child_process', 'aws-sdk' and others

iam-adty, 2017-10-04:
I got these errors when building my react js project.

`./~/escpos/adapter/network.js`
`Module not found: Error: Can't resolve 'net' in '/home/adty/Projects/klopos/web/myklopos/dev/node_modules/escpos/adapter'`
` @ ./~/escpos/adapter/network.js 2:22-36`
` @ ./~/escpos/index.js`
` @ ./src/js/pages/Report/Sales/BranchSales.js`
` @ ./src/js/config/routes.js`
` @ ./src/js/client.js`
` @ multi (webpack)-dev-server/client?http://192.168.1.19:8080 webpack/hot/dev-server ./js/client.js`

`/~/usb/~/node-pre-gyp/~/request/~/tunnel-agent/index.js`
`Module not found: Error: Can't resolve 'tls' in '/home/adty/Projects/klopos/web/myklopos/dev/node_modules/usb/node_modules/node-pre-gyp/node_modules/request/node_modules/tunnel-agent'`
` @ ./~/usb/~/node-pre-gyp/~/request/~/tunnel-agent/index.js 4:10-24`
` @ ./~/usb/~/node-pre-gyp/~/request/lib/tunnel.js`
` @ ./~/usb/~/node-pre-gyp/~/request/request.js`
` @ ./~/usb/~/node-pre-gyp/~/request/index.js`
` @ ./~/usb/~/node-pre-gyp/lib/install.js`
` @ ./~/usb/~/node-pre-gyp/lib ^\.\/.*$`
` @ ./~/usb/~/node-pre-gyp/lib/node-pre-gyp.js`
` @ ./~/usb/usb.js`
` @ ./~/escpos/adapter/usb.js`
` @ ./~/escpos/index.js`
` @ ./src/js/pages/Report/Sales/BranchSales.js`
` @ ./src/js/config/routes.js`
` @ ./src/js/client.js`
` @ multi (webpack)-dev-server/client?http://192.168.1.19:8080 webpack/hot/dev-server ./js/client.js`

`./~/serialport/lib/bindings/unix-read.js`
`Module not found: Error: Can't resolve 'fs' in '/home/adty/Projects/klopos/web/myklopos/dev/node_modules/serialport/lib/bindings'`
` @ ./~/serialport/lib/bindings/unix-read.js 2:11-24`
` @ ./~/serialport/lib/bindings/darwin.js`
` @ ./~/serialport/lib/bindings/auto-detect.js`
` @ ./~/serialport/lib/index.js`
` @ ./~/escpos/adapter/serial.js`
` @ ./~/escpos/index.js`
` @ ./src/js/pages/Report/Sales/BranchSales.js`
` @ ./src/js/config/routes.js`
` @ ./src/js/client.js`
` @ multi (webpack)-dev-server/client?http://192.168.1.19:8080 webpack/hot/dev-server ./js/client.js`

i load the module with `require` after `import`

`import React from "react";`
`var escpos = require('escpos');`

what are the errors about and how to fix them?

adenvt, 2017-10-12:
escpos only support for node / electron application... not support for browser...

if you actually using node / electron, don't forget to set `target: 'node'` on your webpack config...

closed 2017-10-12 by song940

iam-adty, 2017-10-13:
Yap. Thanks man. i've got it work by build it to an electron application.