lsongdev/node-escpos #81: Does it work with STAR TSP143lll LAN thermal printer ?

ax88, 2017-10-07:
It works without any issue with a Epson TM-T20 II Ethernet printer.
Using the same sample code, It doesn't work with a STAR TSP143lll LAN.

JefferyHus, 2017-12-11:
I confirm facing the same problem now, STAR printers aren't working

Psychopoulet, 2018-02-15:
are you sure that STAR printers (or just this one) support escpos standard ?
I thought it was created by EPSO, so maybe all the printers does not reconize this standard ?

JefferyHus, 2018-02-15:
I have a different STAR printer and it didn't work as well. So I assume either all STAR printers or a series of it.

Psychopoulet, 2018-02-16:
http://www.starmicronics.com/absolutefm/absolutefm/afmviewfaq.aspx?faqid=72

JefferyHus, 2018-02-16:
It is activated as I have created a buffer and sent it out to print, and it worked.

JefferyHus, 2018-02-16:
I investigated the issue, it's because of the buffer bits that my STAR printer accepts it's different than what is present in this library so far, so when I used its manual, things worked fine with me.

ax88, 2018-02-16:
How to change the buffer bits in the star printer ? And what value did you put ?

JefferyHus, 2018-02-17:
I followed the manuals mate, check for your printers manual, there should be a part of commands then use those commands in your buffer.

jayel74, 2023-10-18:
> created

Hello,

I am struggling with this, I have a TSP100iii and it doens't print anything. Can you explain what the solution is?
You created a buffer? What does that mean?

greets John

Psychopoulet, 2023-10-19:
is escpos enabled on your printer ?

jayel74, 2023-10-19:
Yes it is enabled

Psychopoulet, 2023-10-20:
Are you sur the connection is done ? what are you using, serial, network, bluetooth, usb ?

jayel74, 2023-10-22:
`'use strict';
const escpos = require('escpos');
escpos.Network = require('escpos-network');

//escpos.SerialPort = require('escpos-serialport')
// const device  = new escpos.USB(0x0416, 0x5011);
// const device  = new escpos.RawBT();
// const device  = new escpos.USB(0x0416, 0x5011);
// const device  = new escpos.RawBT();
const device  = new escpos.Network('192.168.18.118');
// const device  = new escpos.Serial('/dev/usb/lp0');
const printer = new escpos.Printer(device);

device.open(function(error){
    if (error){
        console.log(error)
        return
    }
    printer
        .font('a')
        .align('ct')
        .style('bu')
        .size(1, 1)
        .text('The quick brown fox jumps over the lazy dog')
        .text('敏捷的棕色狐狸跳过懒狗')
        .barcode('1234567', 'EAN8')
        .table(["One", "Two", "Three"])
        .tableCustom(
            [
                { text:"Left", align:"LEFT", width:0.33, style: 'B' },
                { text:"Center", align:"CENTER", width:0.33},
                { text:"Right", align:"RIGHT", width:0.33 }
            ],
            { encoding: 'cp857', size: [1, 1] } // Optional
        )
        .qrimage('https://github.com/song940/node-escpos', function(err){
            this.cut();
            this.close();
        });

});`

Psychopoulet, 2023-10-23:
no error, to timeout ? tried a console.log on the last callback to show if an error occured ?