lsongdev/node-escpos #85: Can't close device with a pending request

bahrul221, 2017-10-24:
this is my print function
<pre>print() {
    const device = new escpos.USB(0x04B8, 0x0202);
    const printer = new escpos.Printer(device);
    device.open(function () {
      printer
      .font('a')
      .align('ct')
      .text('Hello World')
      .cut()
      .close(() => {
        console.log('close');
      });
    });
  }</pre>

close's callback never called.
<pre>.close(() => {
       console.log('close');
});</pre>

The first time calling this function, the printer works, and next calling throw this error

<pre>Error: Can't close device with a pending request
    at Device.usb.Device.close (E:\workspace\tester\electron\escpos\dist\node_modules\usb\usb.js:43:7)
    at USB.close (E:\workspace\tester\electron\escpos\dist\node_modules\escpos\adapter\usb.js:127:15)
    at OutEndpoint.<anonymous> (E:\workspace\tester\electron\escpos\dist\node_modules\escpos\printer.js:454:18)
    at callback (E:\workspace\tester\electron\escpos\dist\node_modules\usb\usb.js:362:14)
    at E:\workspace\tester\electron\escpos\dist\node_modules\usb\usb.js:368:33
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickCallback (internal/process/next_tick.js:104:9)
</pre>

Thanks

Psychopoulet, 2017-10-25:
if you don't send endofline with "println" function, try some "flush" function, like

      .cut()
      .flush()
      .close(() => {
        console.log('close');
      });

bahrul221, 2017-10-25:
I think it's because `flush()` was called twice from `cut()` and `close()`. 
So I modified this at printer.js, then it works
<pre>
Printer.prototype.cut = function(part, feed){
  this.feed(feed || 3);
  this.buffer.write(_.PAPER[
    part ? 'PAPER_PART_CUT' : 'PAPER_FULL_CUT'
  ]);
  // return this.flush();
  return this;
};
</pre>

closed 2017-10-31 by song940