lsongdev/node-escpos #86: Twice Print

VinZenTz1989, 2017-10-24:
```
'use strict';
const escpos = require('escpox');
const device  = new escpos.Serial('COM6');
const printer = new escpos.Printer(device);

device.open(function(err){
  printer
  .font('a')
  .align('ct')
  .text('The quick brown fox jumps over the lazy dog')
});
```
my printer is STAR TUP900, 
I use this code for testing, but every I run this code. it prints twice time.

first paper doesn't have any content;
but second paper has a text that write in code 

Any Suggestion?

Psychopoulet, 2017-10-25:
don't forget to use the "close" function ?

song940, 2017-10-25:
and don't foget `.cut()` .

VinZenTz1989, 2017-10-25:
I tried it but still no content paper when running command

VinZenTz1989, 2017-11-01:
Now, I found cause of the issues

```
Printer.prototype.cut = function(part, feed){
  this.feed(feed || 3); // it's working when I commented this line. Is it a bug? 
  this.buffer.write(_.PAPER[
    part ? 'PAPER_PART_CUT' : 'PAPER_FULL_CUT'
  ]);
  return this.flush();
};
```

closed 2017-11-06 by VinZenTz1989

rafaelvaloto, 2018-03-28:
I know this is already closed but I am responding if someone else needs an answer

This cut parameter (PAPER_PART_CUT) makes two prints, and cut (PAPER_FULL_CUT) cuts the whole paper only makes one print, I do not know if it's correct but I used the PAPER_CUT_A settings for the first print and PAPER_CUT_B for the second print and then.