lsongdev/node-escpos #88: Any specification for other brand POS printer ? My printer work only English character.

MonkeyDoll, 2017-11-01:
Hello,
I'm using GPrinter / Model GP-58MB which is supported ESC/POS commands. And supported Chinese, Korea, etc. languages.
You can refer. to here http://www.gprinter.hk/pos-printers-58mb.html
And i try brought a printer to test on real POS system that can print receipt multiple language and work fine.

But from sample only English characters printout other will be print as unreadable characters.

MacOS Sierra 10.12.6
NodeJS v8.9.0
[image]

fanchou, 2017-11-07:
@MonkeyDoll Maybe your code clashes with baudRate of printer

uyu423, 2017-11-20:
please check character encoding set

MonkeyDoll, 2017-11-21:
@fanchou If that is the case, how to solve baudRate issue ?

@uyu423 Do you mean encoding on printer/hardware's setting ? Or script level ? 
If it's a script level's issue, I using sample code it mean that sample code has issue right ?

fanchou, 2017-11-22:
@MonkeyDoll Read the instruction，how to change baudRate.  the printer >= your code set(9600)

[image]

Psychopoulet, 2018-02-15:
any question now ?

komikoni, 2018-02-17:
When not in kanji mode,
It is necessary to execute the start command of kanji mode.
(In most cases, once you run it, the state of kanji mode will be remembered)

```javascript
// Select Kanji character mode
await printer.print('\x1c\x2e') // FS .
// Cancel Kanji character mode
await printer.print('\x1c\x26') // FS &
```

monajafi, 2018-03-03:
I have the same issue with BeiYANG BTP-2002NP model the printer just prints english characters
what is the solution around this

Psychopoulet, 2018-03-06:
you can test all the encodings until you found the good  one.
are you sure that your printers supports kanjis ? (... kata, katakana, cyrillic, whatever you want)

monajafi, 2018-03-07:
I ended up delegating print job to C# console code. this library's print speed is lower than direct print

song940, 2019-03-04:
Any questions?

closed 2019-03-04 by song940