lsongdev/node-escpos #89: Can't Install

rafael-nascimento, 2017-11-06:
Hi! How are you doing?

I'm having a hard time trying install escpos via npm.
I'm getting error a lot of error. And it tells me to try and run it as adminstratior.
I tried opening git bash/cmd as adminstrator, cleaned npm cache, but nothing worked so far.

Do you have any clue of what is happening?

Thank you in advance.

rafael-nascimento, 2017-11-06:
Solved this issue downgrading node version. Thanks anyways!

closed 2017-11-06 by rafael-nascimento

miguelrcDEV, 2017-11-20:
Which version of node installe @rafael-nascimento ? I have the same error with node 8.9.0

Psychopoulet, 2017-11-21:
if you have a failure with libusb's compilation (most common errors with node >= 8), here is the process (on windows)

npm install iconv-lite mutable-buffer get-pixels qr-image && _install_escpos.bat && cd %~dp0 && _install_usb.bat && cd %~dp0 && cd %~dp0\node_modules && git clone https://github.com/song940/node-escpos/ --recursive move node-escpos escpos && cd %~dp0\node_modules && git clone https://github.com/tessel/node-usb --recursive && cd node-usb/libusb && git remote add upstream https://github.com/libusb/libusb.git && git fetch upstream && git checkout upstream/master && cd .. && npm install && cd .. && move node-usb usb

miguelrcDEV, 2017-11-22:
This give me an error in the second nested command: 
"_install_escpos.bat" it is not recognized as an internal or external command,
program or batch file executable.

Psychopoulet, 2017-11-22:
sorry, forgot this. edited.

Psychopoulet, 2017-11-22:
the point is, you have to install the module usb without the libusb dependendy, install the last version of libusb (which is not updated in the node package), then compile it

miguelrcDEV, 2017-11-22:
Ok. I re-run the edited command but still having error with <code>cd %~dp0</code>, but i try later to do manually installation of usb without libusb dependency on windows as you say.

Psychopoulet, 2017-11-23:
cd %~dp0 is for scripts (forgot to translate), you can adapt them