lsongdev/node-escpos #95: open cashdraw can't not effect the first time without print another thing

LorneZH, 2017-12-05:
exports a function ：
exports.openCashDraw = function(dir,res,content){ 
	try{
		device.open(function(){
			  printer
			  .cashdraw()
	  		  .close();
		});
		/*device.open(function(){
		  printer.cashdraw()
		});*/
	}catch(e){
		content = {
		  	code:998,
		    message: "钱箱设备未连接或未开启"
		};
	}
	if(content == undefined){
		content = {
		  	code:200,
		    message: "开箱成功"
		};
	}
	res.send(content);	
}

opencashdraw not immediately be effected，and affected when i used this function （device.open） second time

Psychopoulet, 2018-02-15:
try to use it in a asynchronous way (promise ? await ? callbacks ?)

LorneZH, 2018-03-16:
using other method after cashdraw() and the cashdraw open success~

		device.open(function(){
			  printer
			  .style('bu')
			  .cashdraw()
			  .font('a')
	  		  .close();
		});

Psychopoulet, 2018-03-16:
flush problem. as i said, try to use it in an asynchronous way

new Promise((resolve, reject) => {

	  device.open(() => {
		  printer
		  .style('bu')
		  .cashdraw()
		  .font('a')
  		  .close((err) => {
			return err ? reject(err) : resolve();
		  });
	});

});