lsongdev/node-escpos #97: Euro symbol

phate82, 2017-12-13:
Hi,
what encode parameter should I use to print the Euro symbol (€)?
Thanks

Psychopoulet, 2017-12-13:
try cp863 ?

phate82, 2017-12-13:
Hi,

thanks for your answer. I tried cp863, but a question mark is printed instead of € sign.
Any idea? Thanks again

Best regards.

Psychopoulet, 2017-12-14:
verify your printer's encoding, it may not support this one
I use cp863, which works fine with € and french accents

song940, 2018-01-10:
@phate82 any questions right now ?

closed 2018-02-15 by Psychopoulet

Elytum, 2018-08-15:
Hi ! It appears that I'm not the only one with that problem. I tested both Cp863 and Cp858 and none works. By the way Cp863 does include the symbol (https://en.wikipedia.org/wiki/Code_page_858 -> 0x20AC) but Cp858 don't (https://en.wikipedia.org/wiki/Code_page_863), are you sure it works despite of that ?

Here's the code I used to generate a ticket (Hex code also, in case the copy/pasted character was wrong):

`printer.text('€', 'Cp863');
printer.text(String.fromCharCode(0x20AC), 'Cp863');
printer.text('€', 'Cp858');
printer.text(String.fromCharCode(0x20AC), 'Cp858');`

And here's the result:
[image]

Any idea why ? Thanks !

Psychopoulet, 2018-08-16:
not "Cp863" but "cp863".
personnaly, I put a global encodate (const printer = new escpos.Printer(device); printer.encode("cp863");) to avoid to set it at each "printer.text"

Elytum, 2018-08-16:
Maj doesn't seem to do anything, I've got the exact same ticket with or without it. And yes, I encode globally but I wrote it inside the printer.text call in order to simplify the copy/pasted code.

Psychopoulet, 2018-08-16:
so your printer does not support this encoding, I guess. try another one, which support €, like utf8

Elytum, 2018-08-16:
The weird thing is that I've got a ticket printed by that printer containing the € character (Using text, not images) and the self-test does list "PC858".

MLDMoritz, 2022-03-11:
CP858 contains EUR Symbol on D5 / 0x20AC. It works fine when printing code pages. The character displayed does not match any char on the code page. The last char (wonky F) of @Elytum is what get's printed at my printer too (TH230):

Theres probably a deeper issue? Did you get it resolved?