lsongdev/node-escpos #98: Font Size when it prints Chinese Char

rogerlinchehao, 2017-12-15:
Hi, I am currently using Epson TM-T82 Printer with this library.

I faced some issue when I am trying to print chinese characters, the font size seems to be out of control. 

Please advise and refer to my code and the printout below.

Thank you

[image]

[image]

komikoni, 2018-02-17:
Hi

Changing the font size is also possible in Chinese.
Of course, the terminal side needs to correspond.

```javascript
const escpos = require('escpos');
const device  = new escpos.USB();
const printer = new escpos.Printer(device);

device.open(() =>{
  printer
  .font('a')
  .align('ct')
  .style('normal')
  .text('Hello World')
  .text('')
  .text('')
  .size(2, 2)
  .text('large')
  .text('敏捷的棕')
  .size(2, 1)
  .text('敏捷的棕')
  .size(1, 1)
  .text('敏捷的棕')
  .font('a')
  .align('lt')
  .style('normal')
  .text('2018/02/17 00:00')
  .cut()
  .close();
});
```

komikoni, 2018-02-17:
The following is the support command of TM-T82.
The size function uses the "ESC!" command.

https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=81

pholmesp, 2018-10-21:
Hi rogerlinchehao,

I am also trapped in printing Chinese words via a thermal printer and my situation is even worse as I haven't yet figured out how to get the words printed properly, so I wonder what character set is used in your code?

Thanks

song940, 2019-03-04:
Any questions?

closed 2019-03-04 by song940