lsongdev/node-escpos #99: What About Beep Sound ?

ozgursel, 2018-01-02:
I am trying to add beep functionalty..

am still searching..

> If it is Epson ESC/POS, you can try sending an ASCII BEL (Hex value 0x07) without any other characters. It is not recommended by Epson, but it might work.

[ESC/POS COMMANDS](https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=72)

song940, 2018-01-10:
em.., My terminal printer does not have buzzer, and ... pull-request are welcome 😄

ozgursel, 2018-03-09:
```
ASCII      ESC   B     n   t
HEX        1B    42    n   t
Decimal    27    66    n   t
```
1<=n<=9,1<=t<=9

• "**n**" Refers to the number of buzzer times.
• "**t**"  Refers to the buzzer beeps every few hours (t * 100) milliseconds.

am on it...

ozgursel, 2018-04-20:
#120

closed 2018-04-20 by ozgursel