lsongdev/node-escpos #106: Image printing - Question and Help

sam-lex, 2018-01-29:
Hi, what's the difference between `.image` and `.raster`?

I'd also like to get help also with image printing quality.

I'm printing like this:

```typescript
  private _printImage(res: express.Response, img: string) {
    try {
      const imgFileName = this._getImgFileName();
      this._saveImgToFile( img, imgFileName );
      escpos.Image.load( imgFileName, (image) => {
        this.printer
          .align('ct')
          .raster( image );
        this.printer.cut();
        res.json('Print success');
      });
    }
    catch (e) {
      console.error('Error on print image: ', e);
      res.sendStatus( 500 );
    }
  }
```

This is the sample image that I used for testing (not really the same though, I generate the receipt from an html canvas, and then save it to a file. I have a mac retina display, so there's the issue with pixel ratios... My client has the printer connected to a windows 32 bits, and I have no printer with me, so this makes it a bit hard to test every possible configuration.. I'll have to investigate it later why it seems to be generating wrong format.)
[image]

And this is the result of printing:
[image]

Psychopoulet, 2018-02-15:
"Hi, what's the difference between .image and .raster?"

it does not use the same format :
- image => BITMAP
- raster => GSV0

bslayton, 2018-06-06:
Not sure if you ever found a solution, but came across this and thought I'd chime in for you or someone else.

There's a possibility that your printer may have a darkness setting, which my understanding basically controls the amount of heat applied to the paper. If so, and it's turned up too high, it would heat the paper too much and give it results like what you have.

While this doesn't appear to be a Zebra printer, i'm sure you can expect similar issues from other thermal printers. It can be a hassle because getting a good thermal print is dependent on paper/label quality, and quality of the thermal print head.

sam-lex, 2018-06-06:
thanks for the info, I have no access to their printer anymore, it was a chinese brand xp-printer. The heat control would probably be the easiest solution.

Based on what you said, my guess is that this printer takes a bit longer to process images, hence the heat "accumulates" as it slowly prints. Because normal text prints out just fine, unfortunately it didn't support the language I needed.

It's been a while, but I think having something like this could be the fix (png2pos): https://github.com/song940/node-escpos/issues/62
if I'm not wrong it is/was only available to linux.

If I ever come up with another project that requires printing, I'll try implementing png2pos in js.