lsongdev/node-escpos #111: Convert bufffer to string

joshuaalpuerto, 2018-02-23:
Hi awesome library!

I have a question if its possible to convert the `buffer` to `string`. I have a use-case that I need to send the string through an API.

Thanks!

Psychopoulet, 2018-02-23:
?? check the node api documentation ?
https://nodejs.org/api/buffer.html#buffer_buf_tostring_encoding_start_end

joshuaalpuerto, 2018-02-23:
@Psychopoulet 

Thanks for the reply
I'm actually using this code to get the string from buffer.
```js
printer
    .font("a")
    .align("ct")
    .text("The quick brown fox jumps over the lazy dog")
    const buffer = printer.buffer.join().toString('utf8')
```
I am able to produce the string, However when submitting this to the API some of the parts are not printed. So Im not sure if Im converting it correctly, or there's something wrong with the device.

Device: **Posiflex**
http://www.posiflex.com/en-global/product/index/POS-Printer

Cheers!

Psychopoulet, 2018-02-23:
If you want to print the sended string, without format encoding, use "utf8" encoding
if you want to print the exact data sended (transformed in bits), you have to use "hex" encoding

Psychopoulet, 2018-02-23:
furthermore, pay attention to your printer standards : are you sure it support escpos ? (if yes, is it activated ?)

joshuaalpuerto, 2018-02-23:
I tried switching from `utf8` to `hex`.

It was read by the printer, howerver the printed item was all number

```js
request({
      url: 'http://<ipaddress>/print',
      method: 'POST',
      headers: {
        'Authorization': '202..'
      },
      body: buffer.toString('hex')
    }, (error, response, body) => {
      if (!error && response.statusCode == 200) {
        res.send('ok')
      } else {
        console.log('Error!', error, response)
      }
    })
```

joshuaalpuerto, 2018-02-23:
> (if yes, is it activated ?)

Yes since its already used by kiosks

Psychopoulet, 2018-02-26:
of course it's numbers ! it's hexadecimal numbers.
it's what your printer received, interpreted, and printed

joshuaalpuerto, 2018-02-26:
@Psychopoulet

Yeah, notice that. :) Anyway It seems that the printer was manipulated into only getting the hex directly into string without conversion. 

```js
const template = `\x1b\x40${String.fromCharCode(27)}${String.fromCharCode(97)}${String.fromCharCode(49)}\n\n\n\x1b\x45\x017-CONNECT`
```

Thanks for your guidance!

closed 2018-02-26 by joshuaalpuerto