lsongdev/node-escpos #124: Problem with center image.

GuilhermeReda, 2018-04-26:
```javascript
const escpos = require('escpos');
const device = new escpos.USB(parseInt('FFF0', 16), parseInt('0100', 16));
const printer = new escpos.Printer(device);

escpos.Image.load(__dirname + '/tux.png', function (image) {
  device.open(function () {
    printer
      .align('ct')
      .image(image)
      .cut();
  });
});
```

[image]

mrarticuno, 2018-04-30:
Fala guilherme beleza ? rapaz eu resolvi esse problema com esse codigo:

`escpos.Image.load(dirImage, function(img){
                    window.device.open(function(){
                        window.printer
                            .font('b')
                            .align('ct')
                            .size(1, 1)
                            .text('\n')//48
                            .raster(img, 'dwdh').close();
                     });
});`

A proposito cuidado se estiver usando impressoras EPSON, elas costumam não funcionar o print de imagens.

GuilhermeReda, 2018-04-30:
This is the output.
[image]

[999999999.zip](https://github.com/song940/node-escpos/files/1960877/999999999.zip)

mrarticuno, 2018-04-30:
Try this code and see if any of theese cmd work as intend btw

For PT-BR printing use 
const options = { encoding: "CP860" };// Allows you to use Special Characters as Ç ã á and so on.
const printer = new escpos.Printer(device, options);

printer
    .align('ct')

    .image(image, 's8')
    .image(image, 'd8')
    .image(image, 's24')
    .image(image, 'd24')
    
    .raster(image)
    .raster(image, 'dw')
    .raster(image, 'dh')
    .raster(image, 'dwdh')

    .cut();

mrarticuno, 2018-04-30:
Btw, why you're not using the qrImage function from the escpos ?

GuilhermeReda, 2018-04-30:
only the .image one works
all the raster give me that output, and qrImage use raster.

mrarticuno, 2018-04-30:
@GuilhermeReda R u using Zadig drivers ? probably after build your application will not work as intend.

GuilhermeReda, 2018-04-30:
using zadig drivers.
I think it's a problem to this printer. it's very limited.

mrarticuno, 2018-04-30:
Probably, as I could see in the manual the commands are very limited but
you can change the qrImage to Bitmap instead of raster, try read
the qr-image npm repo

best of lucky for you.

closed 2021-11-08 by GuilhermeReda