lsongdev/node-escpos #126: "ERROR TypeError: Buffer is not a constructor" when trying to create printer object

u2633, 2018-05-03:
[image]

node version: 8.11.1
npm version : 5.6.0

Is there any one who has the same issue like me!?

Psychopoulet, 2018-05-03:
normal. new Buffer() must not be used, it's deprecated. we should use Buffer.from() instead.

Psychopoulet, 2018-05-03:
https://github.com/song940/node-escpos/pull/127

u2633, 2018-05-03:
👍

u2633, 2018-05-03:
@Psychopoulet 
Thanks. In my case, "new Buffer()" can still work in my develop build, but failed 
 on production build.

song940, 2018-05-03:
emm, We depend a package named [mutable-buffer](https://www.npmjs.com/package/mutable-buffer) and that publish a Incompatible version ...

song940, 2018-05-03:
Have you log "Buffer" object in console ?
and  mutable-buffer version ?

u2633, 2018-05-03:
@song940 thanks for your quick reply.

yes, I have tried to log that, the result as below:
[image]

I found that mutable-buffer version is still 2.1.1 on your project, but I have tried to override the version to 2.0.3 in my pakages.json. 

I'm using electron and angular these two frameworks and using angular CLI as the compiler then packages my project by using [electron-builder](https://github.com/electron-userland/electron-builder)

Psychopoulet, 2018-05-03:
mutable-buffer...
woops, sorry guys, not enougth coffe I guess ><

song940, 2018-05-03:
@u2633 i tried fix that issue again, it works on my terminal printer .

closed 2018-05-03 by song940

u2633, 2018-05-04:
Thanks guys, I also debug with my project and It finally works! I found something interesting about buffer flush. I used to call cut() method at the end of print process, and it would do flush before, however I have to use flush() at present. Anyway, it is really such a good experience to me.

commit 8903834e (2018-05-03): fix #126
commit 65122be3 (2018-05-03): fix #126 again ...