lsongdev/node-escpos #129: cordova plugin please ?

gurselk, 2018-05-03:
Is there a plugin that can work with cordova?

Psychopoulet, 2018-05-03:
try this one, if it doesn't search on google

closed 2018-05-03 by Psychopoulet

jor3l, 2018-05-03:
It depends on what you want to achieve, if is a bluetooth printer you have to use the bleno plug in or similar then write to the device just as this plugin does

TonyLuo, 2018-06-22:
I would like to use it in ionic project with cordova Bluetooth Serial to connect to bluetooth printer , and get following error. 

what I need is the Printer to generate byte[] and then send the byte to bluetooth printer, is it possible to make the Printer much more independence?

```
index.js:3 Uncaught TypeError: Cannot read property '_handle' of undefined
    at index.js:3
    at Array.forEach (<anonymous>)
    at module.exports (index.js:2)
    at Object.<anonymous> (log.js:11)
    at Object.<anonymous> (log.js:309)
    at __webpack_require__ (bootstrap 664a95231212dba0ad99:54)
    at Object.<anonymous> (node-pre-gyp.js:16)
    at Object.freeExports (node-pre-gyp.js:202)
    at __webpack_require__ (bootstrap 664a95231212dba0ad99:54)
    at Object.<anonymous> (usb.js:1)
```

Ng-India99, 2019-08-09:
Hi @TonyLuo  did find the solution for print in ionic ?

TonyLuo, 2019-08-09:
i just removed the usb.js releated code

Ng-India99, 2019-08-09:
hi @TonyLuo  how we can delete usb.js file becouse that is stored on plugin ?

iam getting following errors when i installed this plugin in blank project.

client:159 ./node_modules/escpos/node_modules/@serialport/bindings/lib/linux-list.js
Module not found: Error: Can't resolve 'child_process' in 'C:\Users\Surendra\testPrinter\node_modules\escpos\node_modules\@serialport\bindings\lib'