lsongdev/node-escpos #130: encode arabic letters

geekly88, 2018-05-22:
i try to print arabic letters but it always gives me unknown results

like question marks or other depending on type of encode that i pass to .encode function

i try UTF-8 and different types of char encodes

the Code: 

'use strict';
var escpos = require('escpos');
var iconv = require('iconv-lite');
var device  = new escpos.USB();
var options = { encoding: "UTF-8" /* default */ }
const printer = new escpos.Printer(device,options);
device.open(function(err){
  if(err) throw err;
  printer.model('qsprinter')
  .font('a')
  .align('RT')
  .size(1, 1)
  .text('تجربة الترميز')
  .text(iconv.encode('تجربة الترميز' , 'UTF-8'))
  .cut()
  .close()
});

eagle-design, 2019-06-09:
Did you solve it ?

AhmedSobh, 2020-07-07:
@song940 can you help us in the Arabic encoding issue?