lsongdev/node-escpos #136: Setting printer encoding from escpos commands, is that possible?

mreis1, 2018-06-26:
I've been looking for a escpos command to set the printer's encoding page. 
So far all I managed to do is use the manual way. In my EPSON TM-20 I powered off the printer, switch it back on while pressing and holding the feed button, then following the printed instruction to do the setup.

yhauxell, 2018-06-26:
@mreis1 not sure if this will help you in my case I have to print latin characters like ñ or á for example, so what I do is to set the code table to LATIN1 and encoding to iso-8859-1. if you need some help on that please let me know.

mreis1, 2018-06-26:
Hello @yhauxell. I think I found a PDF that contains a complete list of commands and some of them must be those I'm looking for. Are you using the same commands to set your printer to LATIN1 and encoding to iso-8859-1? I can't try them right now, but I'll do that tomorrow.

For reference I leave the link to the [PDF here](https://files.support.epson.com/pdf/general/escp2ref.pdf)

And here's the most important pages relating to the subject of my question.
`
A-2 	        Character Tables   
C-2		Command List by Function   

C-73 	Assign character table  								ESC ( t     pag 82
C-77	Select character Tables  							ESC t       pag 84
C-80	Select an international character set 	                                 ESC R       pag 87

`

mreis1, 2018-06-28:
@yhauxell I just realize my mistake. 
I was passing a utf8 string directly to the printer. Of course this couldn't work without first converting utf8 characters to PC858 which is my printer's current page settings. 

To set the printer into PC858 I had to execute the following ESCPOS command:
ESC t n 
where I had to replace n with my charset page (19 = PC858)

Decimal 19  = 0x13
So the final hex string command looks like this -> **1b7413**

Then I converted this table to PC858 hex.

```
€£@!éèáà"#$%&'()*+,-./
0123456789:;<=>?
@ABCDEFGHIJKLMNO
PQRSTUVWXYZ[\]^_
\`abcdefghijklmno
pqrstuvwxyz{|}~▒
◄▲▼►╝╗╔╚╣╩╦╠═║╬♦
↑▀▄↓┘┐┌└┤┴┬├─│┼█
 ÀÂÈÊËÎÏ´\`ˆ¨˜ÙÛ₤
¯Ýý˚ÇçÑñ¡¿¤£¥§ƒ¢
âêôûáéóúàèòùäëöü
ÅîØÆåíøæÄìÖÜÉïßÔ
ÁÃãÐðÍÌÓÒÕõŠšÚŸÿ
Þþ·µ¶¾-¼½ªº«■»±
```

Which produces the following PC858 Hex string
```
d59c4021828aa08522232425262728292a2b2c2d2e2f0a303132333435363738393a3b3c3d3e3f0a404142434445464748494a4b4c4d4e4f0a505152535455565758595a5b5d5e5f0a606162636465666768696a6b6c6d6e6f0a707172737475767778797a7b7c7d7eb10abcbbc9c8b9cacbcccdbace0adfdcd9bfdac0b4c1c2c3c4b3c5db0a20b7b6d4d2d3d7d8ef60f9ebea0aeeedec8087a5a4ada8cf9cbef59fbd0a83889396a082a2a3858a9597848994810a8f8c9d9286a19b918e8d999a908be1e20ab5c7c6d1d0d6dee0e3e5e4e9980ae8e7fae6f4f32dacaba6a7aefeaff1
```

Then all worked just fine. 
Here's another PC that might be useful to someone else
> http://www.starmicronics.com/support/mannualfolder/escpos_cm_en.pdf
> Page 76

yhauxell, 2018-06-29:
@mreis1 Glad to hear you solved your issue, I made my fixes some time ago so I don't remember exactly the commands :( but you can check my fork in any case and see if there is something you can use from there. I believe are very similar since I remember having used the book you mentioned. 

[printer.js](https://github.com/yhauxell/node-escpos/blob/fce91d2371e73a37c3fb42ac64b81504a0210e27/printer.js#L28)
[commands.js](https://github.com/yhauxell/node-escpos/blob/fce91d2371e73a37c3fb42ac64b81504a0210e27/commands.js#L302)

Hope it helps you. Regards

imran-razak, 2019-08-26:
@mreis1 can you please explain what this means "Then I converted this table to PC858 hex.".
How can I do this?