lsongdev/node-escpos #138: QUESTION Does this work with usb to parallel adapters ?

devyaz, 2018-07-02:
I've installed on my system a laptop without a parallel port(obvious on modern laptops) and used an emulator(USB-Parallel) adapter. but it gives an error when i try the examples error is the same for all examples
```
$ node image
C:\Users\Mtawali\Downloads\escpos\node-escpos\printer.js:23
  this.buffer = new Buffer();
                ^

TypeError: Buffer is not a constructor
    at new Printer (C:\Users\Mtawali\Downloads\escpos\node-escpos\printer
.js:23:17)
    at Object.<anonymous> (C:\Users\Mtawali\Downloads\escpos\node-escpos\
examples\image.js:5:17)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:188:16)
    at bootstrap_node.js:609:3

```

song940, 2018-07-03:
what is your nodejs and escpos version?
it’s seems to old, upgrade to fix that.

Psychopoulet, 2018-07-03:
it's probably an old version of escpos which use Buffer (not a constructor) and not MutableBuffer (correct constructor), with a recent version of node (new Buffer was deprecated on old versions, not forbidden)

closed 2018-07-03 by song940

devyaz, 2018-07-03:
Thanks guys it was a problem with driver