lsongdev/node-escpos #139: Doesn't work in Electron

devyaz, 2018-07-04:
in an Electron App if i require  ``` const escpos = require('escpos')```  an error is thrown as in issue #114 
i tried including it in my server code its the same story, i even tried it in my vue pos app new error saying package.json cant be found ,
whats the proper way of including/using this??

Psychopoulet, 2018-07-05:
sorry, I didn't answer to the previous thread because there was no question there, only facts (...and I don't read portuguese).
personnaly, I does not use Electron, and I don't know for the others
... if you find a solution, PR are welcome ^^

yhauxell, 2018-07-05:
Hey, @devyaz in my case I use this module in my electron Angular 5 based application without issues. But I don't think is the framework but the way we import the module, first check that this module exists in your node_modules, then try a different way to import if you are using ES6(not familiar with Vue, sorry):

import {Printer, USB} from 'escpos';
or
import * as Escpos from 'escpos';

*be aware that you may problably rebuild usb and serial to you node version with node-gyp

Hope this help or at least you get a fresh point of view

devyaz, 2018-07-05:
@yhauxell  I'll try that now

devyaz, 2018-07-05:
well it din't work, its now asking for a missing dependency 'aws-sdk'

mrarticuno, 2018-07-06:
Hello davyaz, my project uses Vue and Electron and work's fine, have you tried run "electron-builder install-app-deps" in your project ?

devyaz, 2018-07-13:
Hi Guys in linux i used  ```yarn install``` it works now am yet to do the same in windows where the problem started! am using it in my server code so that it prints from server because i didn't want it interfering with my front end code. here is how i include it
``` const escpos =require ('escpos')``` Now how do i print next line(formatting to print like a supermarket receipt) from a long string of characters ?

Psychopoulet, 2018-07-16:
out of scope ?

devyaz, 2018-07-16:
@Psychopoulet  in windows this is what #114  says:
[image]
 its also what am experiencing in windows but it works in linux using Electron

Psychopoulet, 2018-07-16:
aw, ok, so the problem is only on electron / windows when you use new line ?

devyaz, 2018-07-17:
@Psychopoulet No when we require escpos like so ```escpos =require('escpos')``` and only in Windows (7 64bit in my case ) in an Electron App

Psychopoulet, 2018-07-17:
so, what's the matter with your "new line" question ?

devyaz, 2018-07-17:
well it's off topic yes but there is no example on that so i was wondering how to do it dynamically, like 6 products after a sell. because if i had a template that will be counter productive as the number of products isn't fixed e.g
```
PrintData(){
var arrayFromApp{ product:productData,product:productData....}
arrayFromApp.foreach( blah blah

);
}
device.open(function(err){

  printer
  .font('a')
  .align('ct')
  .style('bu')
  .size(2, 3)
  .text('My Company Shop')
 .size(1, 1)
.text(PrintData())// the actual Data to print
  .cut();
    .close();
});
```
so easch sale from the app should print on a new line

Psychopoulet, 2018-07-18:
you can parse your text and replace each "\n" by the "printer.println" function (which is not in the doc, BTW)

MakakWasTaken, 2018-08-07:
Did anyone find a solution to this?

devyaz, 2018-08-07:
@MakakWasTaken  i found out that it is working in Linux but not in Windows with ElectronNuxt(vue) but others say it works in an Angular or 'plain vue-electron' App. in Linux not a single problem i can use 'escpos' and usb-barcode-scanner' but they all fail in my windows Enviroment (win7 64bit)

keyiis, 2018-08-21:
@devyaz you can use node-printer for windows(7/8/10),i already used in electron+angular project two years ago.but client pc must install printer dirver.

Psychopoulet, 2018-08-22:
I use "printer" package when FastPrinter does not do the job (A4, printers which does not support escpos, etc...), but it's far slower than node-escpos ><

I personnaly use node-escpos on Windows 7/8/10 (without Electron yet), I just compile it on a 32bits computer and share the generated binary (for compatibility).
maybe there is a problem with windows 7 (&+) on a 64bits version ??

bigpower, 2018-08-28:
> electron .

dyld: lazy symbol binding failed: Symbol not found: __ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeE
  Referenced from: /Users/liudaquan/node/printer4/node_modules/_usb@1.3.3@usb/src/binding/usb_bindings.node
  Expected in: flat namespace

dyld: Symbol not found: __ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeE
  Referenced from: /Users/liudaquan/node/printer4/node_modules/_usb@1.3.3@usb/src/binding/usb_bindings.node
  Expected in: flat namespace

why？？？

daniellizik, 2018-12-18:
related to this probably https://github.com/node-serialport/node-serialport/issues/1672

HipsterBrown, 2018-12-18:
@daniellizik The stack trace looks related to `node-usb`, which [has some issues with Electron 3 and Windows](https://github.com/tessel/node-usb/issues/162). If someone tries to use the Serial class in this package, then there could be an issue with Serialport +  Electron 3 until my patch for `electron-rebuild` is merged and released. 

For node-usb, I need to do some digging into prebuilding the electron bindings.

devyaz, 2018-12-23:
But am on Linux, in windows escpos needs Zadig drivers but the
usb-barcode-scanner doesn't work so I switched to Linux and it used to work
untill recently

song940, 2019-03-04:
https://github.com/song940/node-escpos/issues/2
https://github.com/song940/node-escpos/issues/76

closed 2019-03-04 by song940