lsongdev/node-escpos #142: Any plan to support running on client side ( browser ) using WebUSB

leedstyh, 2018-08-01:
Chrome is supporting WebUSB API and I'm interested in it, so is there any plan making this nice package to support it?

Psychopoulet, 2018-08-02:
nop, cause with usb we use drivers like zadig, so we have to control the machine
this is not made for client-side.

furthermore, you can develop a node microserver with this package and communicate with it with your client, using libraries like socketio. that's the way I use it.

leedstyh, 2018-08-03:
Ok, that make sense, thanks @Psychopoulet

closed 2018-08-03 by leedstyh

Psychopoulet, 2018-08-03:
You're welcome

jor3l, 2019-02-12:
Late but you can make an adapter that uses webUSB, It does work with both windows and osx. Also native bluetooth works with thermal printers

Psychopoulet, 2019-02-12:
yep, but there is now a micro-server, so you can use it without create a new adapter

twf-nikhila, 2019-02-14:
@Psychopoulet what do you mean by micro-server? The problem is how can a cloud-based app use the local printer? At the moment i uses the ESC/POS-PHP package to create a client-side server, which needs to be installed in clients machine and cloud app communicates to it via socket. 

But looking to get rid of client-server installation because it depends on PHP.

Do you mean micro-server build using nodejs which client need to run? What dependencies does this micro-server have?

twf-nikhila, 2019-02-14:
@jor3l can you please explain a bit more on the webUSB architecture you're following to print? How easy it is for client to print using this?

Thank you in advance

twf-nikhila, 2019-02-14:
@jor3l @Psychopoulet what do you think on using this package to create an electron app which can take data from cloud and print it to the connected printer?

Psychopoulet, 2019-02-15:
... that's the way I use it ^^

twf-nikhila, 2019-02-15:
@Psychopoulet which way?

jor3l, 2019-02-15:
Electron works fine too, for the browser with USB I use this (not an adapter)

```js
const USBDevices = (() => {
  const USB_VENDORS = [{
    vendorId: 1046 // chinese thermal
  }, {
    vendorId: 0570 // epson
  }, {
    vendorId: 0x03f8 // epson
  }, {
    vendorId: 0x04b8 // epson
  }];

  const sendBytes = async (device, bytes) => {
    return await device.transferOut(1, bytes);
  }

  const claimInterface = async (device) => {
    for (const config of device.configurations) {
      for (const iface of config.interfaces) {
        if (!iface.claimed) {
          await device.claimInterface(iface.interfaceNumber);
          return true;
        }
      }
    }

    return false;
  }

  const init = async () => {
    console.log('[USB_DEVICES] Init called...');

    window.usbPrinters = [];
    const devices = await navigator.usb.getDevices();

    if (devices.length) {
      if (devices[0].opened === false) {
        await devices[0].open();
      }

      let device = devices[0];

      try {
        await device.open();
        await claimInterface(device);
      } catch (error) {
        console.log('[USB_DEVICES]', error);
        return alertify.error(lang.usb_connection_error);
      }

      // adds it to the devices array
      if (!window.usbPrinters.find(d => d.productId === device.productId)) {
        window.usbPrinters.push(device);
      }

      console.log('[USB_DEVICES] Device claimed', device.productName);
      alertify.success(lang.usb_printer_connected);
    }

    navigator.usb.addEventListener('disconnect', e => {
      console.log('[USB_DEVICES] USB device disconnected', e.device);
      let deviceIndex = window.usbPrinters.findIndex(d => d.productId === e.device.productId);

      window.usbPrinters.splice(deviceIndex, 1);

      alertify.error(lang.usb_printer_disconnected);
    });
  }

  const search = async (any = false) => {
    navigator.usb
      .requestDevice({
        filters: any ? [] : USB_VENDORS
      })
      .then(async (device) => {
        try {
          await device.open();
          await claimInterface(device);
        } catch (error) {
          console.log('[USB_DEVICES]', error);
          return alertify.error(lang.usb_connection_error);
        }

        window.usbPrinters.push(device);

        console.log('[USB_DEVICES] Device claimed', device.productName)
        alertify.success(lang.usb_printer_connected);
      }).catch(error => {
        console.log('[USB_DEVICES]', error);
      })
  }

  const print = async (deviceId, printQueue, callback) => {
    deviceId = parseInt(deviceId);
    console.log('[USB_DEVICES] Print called to deviceId', deviceId);

    let device = window.usbPrinters.find(d => d.productId === deviceId)
    if (device) {
      printQueue.forEach(async print => {
        await sendBytes(device, print)
      });

      try {
        callback();
      } catch (err) {}
    } else {
      console.log('[USB_DEVICES] Printer not found');
    }
  }

  return {
    init,
    search,
    print
  }
})()
```

```js
USBDevices.search()
// then just
USBDevices.print(window.usbDevices[0].productId, [... lines2print as bytearray], cb)
```

twf-nikhila, 2019-02-15:
@jor3l Thank you so much for sharing this. This is awesome. So we can use the webusb to connect with the USB or Bluetooth printer in a cloud pos itself. Right?

For anyone interested, you can get all vendorID from here: http://www.linux-usb.org/usb.ids
More detailed reading: https://developers.google.com/web/updates/2016/03/access-usb-devices-on-the-web

Thanks again @jor3l

XAW98, 2022-09-24:
> 

Hello @Psychopoulet 
I would to know how to implement this functionality to form on client side with no 3rd party apps

Psychopoulet, 2022-09-25:
check the samples, there is one for client-server ^^

XAW98, 2022-09-26:
> check the samples, there is one for client-server ^^

@Psychopoulet, I see thanks
I think if I am getting it correctly, what I need to do is make the service and install it on the client side,  thus making my own service without 3rd party apps, right?

Psychopoulet, 2022-09-26:
yes... and no. you can create a server on a unique computer and connect multiples others on this one, which be able to run multiple printers by a local network, or bluetooth

XAW98, 2022-09-26:
@Psychopoulet, I know what I need to do now,  thanks a lot

Psychopoulet, 2022-09-29:
I suggest you to read the documentation :
https://github.com/song940/node-escpos/blob/v3/packages/usb/README.md

XAW98, 2022-09-29:
> I suggest you to read the documentation : https://github.com/song940/node-escpos/blob/v3/packages/usb/README.md

 @Psychopoulet I have done so,  but not working in anyway, installed the drivers,  tuned the examples not my own code,  but still throw not printer found
Any ideas?

theo-1, 2023-03-23:
@twf-nikhila: for the Chinese brand called X-Printer which isn't in the list (http://[www.linux-usb.org/usb.ids](http://www.linux-usb.org/usb.ids)), is there a generic vendor ID to use?