lsongdev/node-escpos #143: TypeError: this.adapter.write is not a function

Xsmael, 2018-08-02:
I am getting this error when trying basic example on windows 7 through usb port;
Here is my code:

```
const escpos = require('escpos');
 
const device  = new escpos.USB(0x456,0x808);
const options = { encoding: "GB18030" /* default */ };
const printer = new escpos.Printer({device, options}); 
 
device.open(function(){
  printer
  .font('a')
  .align('ct')
  .style('bu')
  .size(1, 1)
  .text('The quick brown fox jumps over the lazy dog')
  .text('敏捷的棕色狐狸跳过懒狗')
  .barcode('1234567', 'EAN8')
  .qrimage('https://github.com/song940/node-escpos', function(err){
    this.cut();
    this.close();
  });
});
```

The Error:

```
events.js:183
      throw er; // Unhandled 'error' event
      ^

TypeError: this.adapter.write is not a function
    at Printer.flush (C:\node\thermalPrint\node_modules\escpos\printer.js:568:16)
    at Printer.close (C:\node\thermalPrint\node_modules\escpos\printer.js:592:15)
    at Printer.<anonymous> (C:\node\thermalPrint\index.js:27:10)
    at C:\node\thermalPrint\node_modules\escpos\printer.js:480:26
    at C:\node\thermalPrint\node_modules\get-pixels\node-pixels.js:22:5
    at exports.PNG.<anonymous> (C:\node\thermalPrint\node_modules\pngjs\lib\png.js:75:7)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at exports.PNG.emit (events.js:211:7)
    at exports.PNG.<anonymous> (C:\node\thermalPrint\node_modules\pngjs\lib\png.js:37:10)
```

my printer settings:
HOIN HOP-H58

from escpos.USB.findPrinter() :
```
[ Device {
    busNumber: 1,
    deviceAddress: 13,
    deviceDescriptor:
     { bLength: 18,
       bDescriptorType: 1,
       bcdUSB: 512,
       bDeviceClass: 0,
       bDeviceSubClass: 0,
       bDeviceProtocol: 0,
       bMaxPacketSize0: 64,
       idVendor: 1110,
       idProduct: 2056,
       bcdDevice: 512,
       iManufacturer: 1,
       iProduct: 2,
       iSerialNumber: 3,
       bNumConfigurations: 1 },
    portNumbers: [ 1 ],
    _configDescriptor:
     { bLength: 9,
       bDescriptorType: 2,
       wTotalLength: 32,
       bNumInterfaces: 1,
       bConfigurationValue: 1,
       iConfiguration: 5,
       bmAttributes: 192,
       bMaxPower: 50,
       extra: <Buffer >,
       interfaces: [Array] } } ]

```

song940, 2018-08-02:
emm ...

```patch
- const printer = new escpos.Printer({device, options});
+ const printer = new escpos.Printer(device, options);
```

Xsmael, 2018-08-02:
Yeah, now the printer reacts, but prints blank paper, nothing written on it

Xsmael, 2018-08-02:
Yes! it works now! Thanks a tons, i wasted so much time looking around should have come straight here!

closed 2018-08-02 by Xsmael

Psychopoulet, 2018-08-02:
o_O rtfm ?
github issues are for... issues, not unread doc :p

Xsmael, 2018-08-03:
@Psychopoulet  Oh really ?   what if you read the "readme"  

you will see there `const printer = new escpos.Printer({ device, options });`  the very line that caused the issue.

Maybe you should update the doc as well.

Psychopoulet, 2018-08-03:
THAT is an issue :p
(And a fu**ing good one !)

SlehMas, 2020-10-31:
I am still getting this issue even with the correct constructor